goodtft/LCD-show #1: 5" screen crashing on pi 3

manz411, 2017-01-11:
I done this both online and offline and every time I use the last command
`sudo ./LCD5-show ` my pi 3 restarts and then hits an error and gets stuck there. The error is "Kernel Panic : VFS :Unable to mount root fs on unknown-block". I cannot for the life of me get it working.

goodtft, 2017-01-14:
which version of Raspbian you used for tested?

anabrownee, 2017-01-14:
same issue here with raspbian 8  - any fix?

manz411, 2017-01-14:
At first I used an image of PIXEL that I had and when that didn't work I downloaded the latest one directly from raspberrypi.org. Both attempts failed so I researched a bit and after a while I followed these instructions http://tech.scargill.net/cheap-pi-screen/ by editing the config.txt file manually instead of cloning it.

Managed to get the screen to show on my triple boot (PIXEL, Retropie, Kodi) but I DID NOT install the touchscreen drivers. I figured it was the drivers that kept showing the Kernel panic error. Someone mentioned that to get rid of that panic error you have to change a certain file, here is his comment.

"I managed to work on with my raspberry Pi3. Touch screen works fine. Once you install don't use sudo ./LCD-show command (otherwise you will get kernel panic), just go to /home/pi/Downloads/LCD-show or whatever your file location. Open that folder and edit "cmdline.txt" file... from "root=/dev/mmcblkop2" - "root=/dev/mmcblkop7" . Then you give commands like dpkg.... and Sudo ./LCD-show. it will reboot and touchscreen will work with raspberry pi 3 (jessie installed via noobs)﻿"

I however could only get raspbian to work but could'nt get my other OS to work at all, and after a reboot that the pi did by itself it was stuck on the kodi screen frozen. Everytime I would start I after that it just went straight there so I started again.

Hope this helps people.

PS. Hopefully the touch drivers will work later on the future. Apart from the stuff around of getting them working, to be able to use the screen is a 2 minute job after OS install.

manz411, 2017-01-14:
Also should mention that in my current setup the USB cannot power the screen so using external power, and when I start it up the screen shows rainbow colours/lines down the screen all pixelated, and after a few seconds it kicks in.

davidsnyders, 2017-01-27:
I have exactly the same on a Pi Zero

fansua4life, 2017-02-01:
@manz411   Did the screen actually worked for you. I followed the instructions and and the pi rebooted successfully. However, when I plugged in the screen just jitters with no display.Can someone please help me ?

manz411, 2017-02-02:
What I did, (and what those links do) is force the display to play at 800x480 resolution, and I got the screen to work like that. My screen is connected through HDMI and powered separately and NOT from the Pi. I'm not using the GPIO's as I couldn't get the touch screen to work.

Manuel

NullYing, 2017-02-15:
I have same questions on pi3

goodtft, 2017-02-24:
You can download the images from wiki:
https://github.com/goodtft/LCD-show/wiki/Images-Download-address

Tom900, 2017-03-08:
Very New to the Raspberry, but I also followed the same feed, now I cannot get past End Kernel Panic!
Since I bought a pre-installed SD I don't know how to get past this point as I have to unplug it for it to then crash all over again. Help Please?

vpidone, 2017-04-22:
PI3 and 5inch display also gets me Kernel panic error (although it does get the text out to the full size of the screen, the screen is locked). Is this the best place to get help with this?

MgBoggy, 2017-05-14:
Have you guys solved ! i also facing rainbow screen on my pi2 and pi3 both.

miguelBra, 2018-02-07:
@manz411 Thank you so much, I've been like two weeks stucked with the kernel panic issue but editing the cmdline.txt completely fixed it.

lavolp3, 2018-02-07:
I have sent a pull request (actually twice) months ago to resolve this issue, which has been mentioned here several times.

Unfortunately this is obviously not a work in progress...

Chip-Tronics, 2019-04-09:
@manz411
Does that work on the Retropie on the Raspberry 3B?

vpidone, 2019-04-09:
Please remove me from your email list.

Thanks

wesleyoufc, 2020-01-22:
> @manz411 Did the screen actually worked for you. I followed the instructions and and the pi rebooted successfully. However, when I plugged in the screen just jitters with no display.Can someone please help me ?

You use to play runescape?