goodtft/LCD-show #3: rpi zero support

githubber001, 2017-02-05:
is it planned to support pi zero ?

goodtft, 2017-02-24:
Yes, tested OK.

dirkteucher, 2017-04-01:
Can you provide more info on the operating system you used to test on the raspberry pi zero?  I followed your instructions for the 5 inch elecrow screen on  (Raspian Jessie 2017-03-02) and (Raspian Jessie 2017-02-16) and the resistive touch is not working . The screen behaves like a LCD monitor just fine but there is no touch capability.

githubber001, 2017-06-25:
Hi, the current version has the follow behaviour on a pi zero after installation according to instruction:
--> terminal still boot on the hdmi screen
--> but grapical display remains black with flashing cursor
--> 3.5 display show only backlight.
any idea ?
--> switch back with sudo ./LCD-hdmi is working fine

installation output: 
pi@raspberrypi:~/LCD-show $ sudo ./LCD35-show
mkdir: das Verzeichnis „/etc/X11/xorg.conf.d“ kann nicht angelegt werden: Die Datei existiert bereits
this is raspberrypi kernel
998
need to update touch configuration (Lese Datenbank ... 116587 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von xserver-xorg-input-evdev_1%3a2.10.3-1_armhf.deb ...
Entpacken von xserver-xorg-input-evdev (1:2.10.3-1) über (1:2.10.3-1) ...
xserver-xorg-input-evdev (1:2.10.3-1) wird eingerichtet ...
Trigger für man-db (2.7.0.2-5) werden verarbeitet ...
reboot

githubber001, 2017-06-26:
... the sollution in this post works fine:
https://github.com/goodtft/LCD-show/issues/17

...only change in boot/config-35.txt the two last lines to:

dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,swapxy=1,pmax=255,xohms=60,xmin=200,xmax=3900,ymin=200,ymax=3900
dtoverlay=tft35a:rotate=90

closed 2017-06-26 by githubber001