goodtft/LCD-show #6: impossibility to switch from LCD to HDMI

Pickyflo, 2017-04-02:
Hi, we have the last raspbian version from march 2017 and a 3'5 LCD touchscreen.
The initialisation worked very well, but, we are not able now to return to the hdmi screen.

We tried the : sudo ./LCD-hdmi
But the answer is : "sudo ./LCD-hdmi : command not found"

If we unplug the LCD the raspberry stops booting in the middle...
You think, we miss something??

Pickyflo, 2017-04-02:
Thank's by advance for any answer :)

OlagZepol, 2018-02-12:
Hi, 
I,m new to Raspbery and By now I have the same issue. How did you solve this.
Thank you.

Pickyflo, 2018-02-13:
For now, i didn’t...
I’ve droped the project

Sorry,

mees87, 2018-03-25:
Did you first go into the LCD-show directory? That was my mistake.

raysuelzer, 2018-09-28:
Same issue. My HDMI port no longer works... The LCD screen is not ideal except for on the go.

Skierman, 2018-10-18:
I had the same issue but this fixed it:
You need to make sure that your LCD-show folder is properly executable.
In the Terminal, navigate to the directory that your LCD-show folder is located in (in my case it was Desktop)
`cd Desktop`
Then you need to run this command:
`chmod -R 755 LCD-show`
then run:
`sudo ./LCD-hdmi`
That should have fixed your issue.