goodtft/LCD-show #10: No touch functionality

seesix, 2017-04-26:
I have followed your installation guide using Raspbian-2017-03-02 on an Pi3b with LCD7C 1024x768 screen.

Both devices are powered from a single 5V  2.5amp power supply, the screen piggybacking power off the Pi via USB

I am able to get the screen up and running but without any Touchscreen functionality.

Anyone have an success trying to get touchscreen functions working with this screen?

goodtft, 2017-04-27:
Hello,
The touch functions Do not need any software,so we think it's a Hardware Problem.
You should make sure The Touch signal should come back to Pi via USB.
So, One end of USB is connected to one of four USB ports on the raspberry pi, and the other is connected to the USB port of the 7-inch LCD module.
and,Please use the USB cable provided in our packing.

another way, you can tested it on computer, for checking the touch functions is ok or not.
2017-04-27 

goodtft

gauone, 2019-04-10:
Hi,

Does the touch function needs a software on a 3.5 inch RPi Display? I also have the screen running without any touch functionality… I downloaded the "Raspbian Stretch with desktop and recommended software" and I used a "dd" command without using SDFormatter or Win32DiskImager for step 1. However, I rigorously followed step 2 and 3.

Thank you for your help

gauone

A124Bob, 2019-06-17:
> I have followed your installation guide using Raspbian-2017-03-02 on an Pi3b with LCD7C 1024x768 screen.
>
> Both devices are powered from a single 5V 2.5amp power supply, the screen piggybacking power [...]

Have the same problem by a joy-it RB-LCD-7-2 Display (also same to Waveshare 7")
LCD is running, but touch is not operating.
Tried X11, Dev-Mode and install driver.
Can't put a USB-Cable between Pi and Display. The display has only Micro-USB for Power and HDMI.

> tar xzvf /boot/LCD-show-160701.tar.gz  
> cd LCD-show/
> chmod +x LCD7C-show
> ./LCD7C-show

A124Bob, 2019-06-17:
Fixed! Touch is running!
I connected the Micro-USB (power and data) of the display to the Pi USB.

dlmiles, 2021-08-14:
**Attempt 1)**  To confirm for me @A124Bob the 7" model touchscreen did not work when:

Micro USB port 1: Attached to RPI 4 (which is powered by its own USB PSU)
Micro USB port 2: Attached to another USB PSU (with current ampere rating capable of powering the RPI 4)

When in this state I get (and **touchscreen not working**):
```
$ lsusb
Bus 001 Device 006: ID 4348:55e0 WinChipHead
```

**Attempt 2)**  When I disconnect that arrangement and make it:

Micro USB port 1: USB lead attached to RPI 4
Micro USB port 2: USB lead attached to RPI 4 (another port)

Now I get in this state (and **touchscreen still not working**):
```
$ lsusb
Bus 001 Device 030: ID 1a86:e2e3 QinHeng Electronics
Bus 001 Device 031: ID 1a86:e2e3 QinHeng Electronics
```
What is happening is in /var/log/Xorg.0.log it keeps finding the USB device twice and ping-pong:
```
 $ grep Class /var/log/Xorg.0.log
[    12.667] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "evdev touchscreen catchall"
[    12.667] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    12.667] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    13.983] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "evdev touchscreen catchall"
[    13.983] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    13.983] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    15.313] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "evdev touchscreen catchall"
[    15.313] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    15.313] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    16.497] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "evdev touchscreen catchall"
[    16.497] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    16.497] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    17.776] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "evdev touchscreen catchall"
[    17.776] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    17.776] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    18.504] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "evdev touchscreen catchall"
[    18.504] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    18.504] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    19.123] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "evdev touchscreen catchall"
[    19.123] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    19.123] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    19.880] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "evdev touchscreen catchall"
[    19.880] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    19.880] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    20.493] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "evdev touchscreen catchall"
[    20.493] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    20.493] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    21.263] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "evdev touchscreen catchall"
[    21.263] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
[    21.263] (**) wch.cn USB2IIC_CTP_CONTROL: Applying InputClass "libinput touchscreen catchall"
```

**Attempt 3)** 

 When I disconnect that arrangement and make it:

Micro USB port 1: USB lead attached to RPI 4
Micro USB port 2: Disconnected

Now I get in this state (and **touchscreen is working**):
```
$ lsusb
Bus 001 Device 003: ID 1a86:e2e3 QinHeng Electronics
```

This information is not exactly CLEAR in the documentation, specially when the device has 2 USB connectors and talks of power usage concerns.

humberry, 2022-01-11:
Some micro usb cables have only power wires and no data wires (necessary for the touch function). I checked it with lsusb, thanks a lot.