goodtft/LCD-show #11: Problem with installation

myckei, 2017-04-27:
Hello.

I have problem with installation my display.
Yeasterday I have made install without problem, but I messed something on my Pi3, so I decided to make clean install.
Download jessie image, write to card, boot and make system resize and other thing.
After that I have downloaded repo, made chmod and went into directory with files.
Now excecute sudo ./LCD32-show command, Pi shutdown, but after reboot, it hangs up in boot process on "systemd[1]: Starting Create Static Device Nodes in /dev...

Display is only white and Pi is doing nothing.

Thx for advice.

Mickey

myckei, 2017-04-27:
Aaaaaaaaaaaaaaah!

Shame on me!!!
I have 3.5" display, not 3.2" :-) After running correct command, everything is OK and Pi is normally running, display shows nice :-)

Sorry for unnecessary topic. 
Solved, meybe better delete this.

Mickey

goodtft, 2017-04-27:
Congratulations to you,
Focus on our driver update site:
https://github.com/goodtft
2017-04-27 

goodtft