goodtft/LCD-show #15: Display stays white

acidwars, 2017-05-16:
I tried to get the display running with your script and several guides I found on the Internet without any changes.

I am using ArchArm and Raspbian - both without success!

Could the display be broken?

masaner1, 2017-05-20:
Hey,
Same here. I bought the display yesterday and the display stays white.
I followed a lot of different guides but nothing changes.

acidwars, 2017-05-21:
I will send the display back because I think its broken... Did you already managed to get it running @masaner1 ?

masaner1, 2017-05-22:
Yes, i tryed the preconfigured image without any updates and the display shows me the raspberry desktop. But i did not find any solution to activate the touchscreen.

acidwars, 2017-05-26:
But it can't be the solution to take that image...

masaner1, 2017-05-26:
Yes, that's right. I think i will send it back too. Having a touchscreen on the raspberry is really nice, but it should work in a normal way.

inakiuy, 2017-06-04:
I have the same issue and found more people reporting same behavour.
https://raspberrypi.stackexchange.com/questions/66504/3-5-kuman-display-stops-working-after-update-upgrade

Maybe this can help:

> Same for me (I'm using the same display). Solved it by re-installing and setting the packages raspberrypi-kernel and raspberrypi-bootloader to "hold":
> 
> sudo apt-mark hold raspberrypi-kernel raspberrypi-bootloader
> before upgrading. It might also work to "revert" to the old kernel/bootloader (instead of reinstalling) by doing:
> 
> sudo apt-get install raspberrypi-kernel=1.2070405-1 raspberrypi-bootloader=1.2070405-1
> but I have not tested this myself as I'm currently just setting up my system. Of course this is not a permanent solution as it will block all future kernel updates, so a real solution would be welcome.

Graveen, 2017-06-07:
Look at my answer here: https://github.com/goodtft/LCD-show/issues/17