goodtft/LCD-show #17: This LCD works when you use a fresh Jessie install, but not after update

Gutspiller, 2017-06-04:
Just bought the screen, and found this out.

I got it to work following the instructions that came with it, but only if it's with a fresh install of 2017-04-10-raspbian-jessie.img

As soon as you run "sudo apt-get update" and "sudo apt-get dist-upgrade" the screen just stays white.

It also ends up killing the Raspbian image as well.

I'm looking for a screen for my Pi, but one where the drivers break simply from performing OS updates is not user, or customer friendly.

inakiuy, 2017-06-04:
I have the same issue and found more people reporting same behavour.
https://raspberrypi.stackexchange.com/questions/66504/3-5-kuman-display-stops-working-after-update-upgrade

Maybe this can help:

> Same for me (I'm using the same display). Solved it by re-installing and setting the packages raspberrypi-kernel and raspberrypi-bootloader to "hold":
> 
> sudo apt-mark hold raspberrypi-kernel raspberrypi-bootloader
> before upgrading. It might also work to "revert" to the old kernel/bootloader (instead of reinstalling) by doing:
> 
> sudo apt-get install raspberrypi-kernel=1.2070405-1 raspberrypi-bootloader=1.2070405-1
> but I have not tested this myself as I'm currently just setting up my system. Of course this is not a permanent solution as it will block all future kernel updates, so a real solution would be welcome.

Graveen, 2017-06-07:
This is not my answer, found on... Amazon.
This worked for me... But the lack of support is concerning.
source: https://www.amazon.com/gp/customer-reviews/R3CGCFPLQ232W2/ref=cm_cr_dp_d_rvw_ttl?ie=UTF8&ASIN=B01CNJVG8K

-----

FURTHER UPDATE: this LCD display also works with the pi zero with no problem at all! (see pic)

UPDATE: this LCD display works just fine with the new Linux 4.4.9-v7+ kernel and May 10, 2016 Raspian release! here are the steps:

1. use this /boot/cmdline.txt:

dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo

2. edit the end of /boot/config.txt to contain:

dtparam=audio=on
dtparam=spi=on
dtoverlay=ads7846,penirq=25,penirq_pull=2,xohms=150,swapxy=1,xmin=300,ymin=700,xmax=3800,ymax=3400,pmax=255
dtoverlay=waveshare35a

3. copy waveshare35a.dtb (found on-line via swkim01's waveshare git hub [use a web search]) to the /boot/overlays directory as:

waveshare35a.dtbo

(note that the new kernel requires a '.dtbo' on overlays now!)

4. edit /usr/share/X11/xorg.conf.d/99-fbturbo.conf to contain:

Section "Device"
Identifier "Allwinner A10/A13 FBDEV"
Driver "fbturbo"
Option "fbdev" "/dev/fb1"

Option "SwapbuffersWait" "true"
EndSection

5. create a file named 99-calibration.conf under /etc/X11/xorg.conf.d to contain:

Section "InputClass"
Identifier "calibration"
MatchProduct "ADS7846 Touchscreen"
Option "Calibration" "3932 300 294 3801"
Option "SwapAxes" "1"
EndSection

6. reboot and enjoy!

folks, there's no real magic here and no need to download a special 'image' of a Raspbian distro... i don't know why vendors make customers jump through insane hoops to get a product like this working... the screen is an XPT2046, aka 'ADS7846,' which is readily supported by the Linux kernel and Raspian releases

this display is a good deal and will provide a lot of fun - why vendors don't make things easier for their customers i'll never know... one of these days, Raspberry Pi users may benefit from vendors providing simple and easy instructions (Adafruit's waveshare approach is convoluted as well and didn't work for me)...

hope this helps someone

Graveen, 2017-06-07:
Dig more: in fact, **everything is working fine** if you ensure the 2 last lines are reversed, so they basically are:

dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,swapxy=1,pmax=255,xohms=60,xmin=200,xmax=3900,ymin=200,ymax=3900
dtoverlay=tft35a:rotate=90

You can simply edit LCD-show/boot/config-35.txt

closed 2018-09-28 by goodtft