goodtft/LCD-show #20: Work on Raspbian lite

rynop, 2017-06-09:
Anyone gotten this to work on Raspbian Jessie Lite (2017-04-10)?  I'd like my pi 3b with the 3.5 LCD to just act as a kiosk (run browser on boot).

Was hoping I could just do the following but have not had luck:

```
sudo raspi-config
#boot options > cli no login
sudo apt-get install -y git xserver-xorg-video-fbturbo openbox midori unclutter xinit
echo "exec openbox-session" > ~/.xinitrc
mkdir -p ~/.config/openbox && cp /etc/xdg/openbox/* ~/.config/openbox

echo "midori -e Fullscreen -a http://www.espn.com" > ~/.config/openbox/autostart

nano ~/.bashrc
#at the bottom put

if [ $(tty) == /dev/tty1 ]; then
   startx -- -nocursor
fi
```

jonathan-reisdorf, 2017-06-10:
It worked for me on latest Raspbian Lite, but I had to add this like to the `/boot/config.txt`:
```
dtoverlay=waveshare35a
```

I also commented out the line starting with `dtoverlay=ads7846` (which shouldn't be necessary if you add your new entry after that one)

jonathan-reisdorf, 2017-06-10:
(for completeness I also need to add that I also installed the `xserver-org xinit lightdm` packages along with my desktop manager, and that I followed all the steps from the readme file of this repo, also including the last lines)

rynop, 2017-06-10:
Thanks I will try that out and report back

jonathan-reisdorf, 2017-06-16:
Also important: install the xinput-calibrator_0.7.5-1_armhf.deb and xserver-xorg-input-evdev_1%3a2.10.3-1_armhf.deb files from this repo, e. g. like this: `sudo apt install ./xinput-calibrator_0.7.5-1_armhf.deb && sudo apt install ./xserver-xorg-input-evdev_1%3a2.10.3-1_armhf.deb`

closed 2018-09-28 by goodtft

almirbi, 2019-01-16:
> (for completeness I need to add that I also installed the `xserver-org xinit lightdm` packages along with my desktop manager

What do you mean by desktop manager? :)

jonathan-reisdorf, 2019-01-16:
I must have meant "window manager", I think I was using i3wm at that time, but shouldn't matter

ForgivenNin, 2019-01-27:
When I tried finding a solution, I was recommended to use putty for it to work.
Hasn't played out yet because it wouldn't install when I typed `sudo install putty -y`.

If the solution(s) above work, will there be a final step-by-step guide (like the first comment)?
I'm just new to Linux and coding as a whole.

ForgivenNin, 2019-01-28:
Thank you for clarifying.

So I continued to try to make it work, but according to the pi, `git` is not a command.

jonathan-reisdorf, 2019-01-29:
You'll need to install git on your pi, typically you'd to that with the `sudo apt-get install git` command