goodtft/LCD-show #21: End Kernel Panic

zwarburg, 2017-06-09:
I followed the exact instructions on the main page and it crashed my device. I now get:

```
End kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,6)
```

inakiuy, 2017-06-10:
this post will help you
https://github.com/goodtft/LCD-show/issues/9

> I had some trouble.
> The installation script not consider the noobs partition schema.
> In the file usr/cmdline.txt i had substitute root=/dev/mmcblk0p2 with /dev/mmcblk0pX where X is the partition number of root created by noobs ( in my case is 7). To check i was start rpi in rescue mode.(shift key on boot). I'm sorry for my english.

zwarburg, 2017-06-10:
@inakiuy Nope didn't help. Tried every value for `x` from 1-9. Nothing works. This destroyed my machine.

kb7453, 2017-06-25:
I don't know how or why, but that worked for me? I just used the number 7 instead of 2 at the end of the root=/dev/mmcblk0p2

randomer666, 2017-07-09:
Solution above worked for me as well, thank you @kb7453

anandkulkarni, 2017-10-13:
Confirming that this works.

mpierre0220, 2017-10-26:
This works for me. Just booted in recovery mode, (Shift key on the Pi  flash screen) changed root=/dev/mmcblk0p2 to root=/dev/mmcblk0p7 in cmdline.txt, rebooted and everything worked fine. This issue is a life saver.

lavolp3, 2017-10-26:
I just wrote a commit to get rid of the issue by copying another
cmdline.txt into boot when NOOBS is used

mpierre0220, 2017-10-26:
Thanks @lavolp3. So if someone clones the repo again and runs LCDXXX-Show, things will now run fine?

closed 2018-09-28 by goodtft