goodtft/LCD-show #23: slow refresh rate/fps on 3.5"

rynop, 2017-06-16:
The refresh rate on my pi 3b running `Raspbian GNU/Linux 8 (jessie)` seems to be pretty low/slow.

How can I increase the refresh rate so I get higher FPS?

My `/boot/config.txt` has:

```
...
dtparam=audio=on
dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,swapxy=1,pmax=255$
dtoverlay=tft35a:rotate=90
```

I have tried increasing `gpu_mem`, it does not have any impact.

jonathan-reisdorf, 2017-06-16:
I think that's a limitation due to the GPIO connection which has its limits - the nice thing is that you can directly connect the screen to the board and don't have to connect via HDMI, but the price is the low refresh rate

goodtft, 2017-06-16:
we Now have new 3.5inch -HDMI Module, it can refresh rate to 100 FPS.
Module Name:MPI3508

rynop, 2017-06-17:
@goodtft do you have a link that gives more details / purchase info? Googling `MPI3508` gets no hits.

vinayprydotinfotech, 2020-07-28:
> The refresh rate on my pi 3b running `Raspbian GNU/Linux 8 (jessie)` seems to be pretty low/slow.
>
> How can I increase the refresh rate so I get higher FPS?
>
> My `/boot/config.txt` has:
>
> ```
> ... [...]

how did you solve this problem Exactly ...How ?  I am facing this problem so pls reply