goodtft/LCD-show #26: PI 3 reboot after LCD28-show command

faisal349, 2017-06-26:
Hi,

I am using waveshare 2.8 inch LCD with PI3. I followed the procedure given in the link below.

https://github.com/goodtft/LCD-show

When I tried to run the **sudo ./LCD28-show** command, my PI got restarted. Everytime I run this command, PI gets restarted. Any ideas?

I am using Raspbian (Jessie-Lite).

Thanks

faisal349, 2017-06-26:
Please find the attached file too.

[image]

goodtft, 2017-07-06:
Now the Driver do not support the newest version Raspbian-07-05,we will find a way update to support.

goodtft, 2018-09-28:
working well in latest driver

closed 2018-09-28 by goodtft