goodtft/LCD-show #27: 7

miozer, 2017-07-12:


miozer, 2017-07-12:
[image]

miozer, 2017-07-13:
i fix the problem after install noobs again and , i opened  terminal  and  put command 
sudo nono /boot/config.txt 
 and add command below at the end of the lines  and it works but now i have  a purple vertical bar of 2-3 pixels on the left side i ll try to fix it soon

"max_usb_current=1"
"hdmi_group=2 hdmi_mode=1 hdmi_mode=87"
"hdmi_cvt 1024 600 60 6 0 0 0"

bmorse526, 2017-09-03:
The lines as indicated are in 5he config file as they are supposed to be.  Boot fails with kernel panic.  Unable to mount file system.  This only happens after issuing the chaos command and loading the driver for the 7B.  Have tried it several times and each time it corrupts the OS.  Requires a complete reinstall.  This display is unusable.

bmorse526, 2017-09-03:
Corrected for spell check corrections.
The lines as indicated are in the config file as they are supposed to be.  Boot fails with kernel panic.  Unable to mount file system.  This only happens after issuing the chmod command and loading the driver for the 7B.  Have tried it several times and each time it corrupts the OS.  Requires a complete reinstall.  This display is unusable.

Xirao, 2017-09-04:
Same problem screen worrking on Windows 10 but not on my raspberry

lavolp3, 2017-09-05:
What OS are you working with? Could it be volumio?

youozhan, 2017-09-21:
I have the same issue when installing the driver for 5inch display. I am reinstalling everything on my raspberry now. Based on the above comments, should I include the change in my config file too?

fencerJP, 2017-12-03:
I have the same issue as OP, running the 7in screen on latest Raspbian installed from NOOBS. I have reinstalled twice, and got the same issue both times on fresh installs. After executing "sudo ./LCD7C-show", I get the same "not syncing: VFS: ..." error as posted above.

fencerJP, 2017-12-03:
Update:
After working through the solutions found above, I was able to get it working by going into the "LCDXX-show" files and changing the following lines.
old:
`sudo cp ./usr/cmdline.txt /boot/`
new:
`if [ -b /dev/mmcblk0p7 ]; then
  sudo cp ./usr/cmdline.txt-noobs /boot/cmdline.txt
else
  sudo cp ./usr/cmdline.txt /boot/
fi`

You then need to create a new file in the /usr/ directory, named "cmdline.txt-noobs" with the following text:
`dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo`

Sorry for formatting problems. You can see an example here:
[LCD7C-show](https://github.com/fencerJP/LCD-show/blob/patch-3/LCD7C-show)
[cmdline.txt-noobs](https://github.com/fencerJP/LCD-show/blob/patch-3/usr/cmdline.txt-noobs)

closed 2018-09-28 by goodtft