goodtft/LCD-show #29: 3.5" Full HD KeDei V1.1 3561

PrinzJuliano, 2017-07-16:
So I got the 3.5" Full HD 3561. Is it covered yet? It uses HDMI and SPI for touch screen. I did not find any drivers yet.

paulcarroll, 2018-05-25:
For what it's worth my kedei 3.5" just sits there with the screen all white. At boot if I leave the hdmi plugged in I get the output (through hdmi, not the screen) up to a point and then nothing (probably about the point it switches from text to graphics mode actually)/
I can ssh in and set it back to hdmi, but the screen itself never seems to work

lavolp3, 2018-05-26:
The reason might be that you use the standard 3.5 inch drivers. These require video output through gpio.
For a kedei screen (I have one too) use the explanation from kedei or use a 3.5 inch HDMI driver from here. You only need a few lines in the config.txt

paulcarroll, 2018-05-26:
Thanks for the reply. I did actually try the "LCD35-show" driver from here and from kedei... and the image with the driver pre-baked from the OEM from osoyoo (and the pre-baked image from kedei for that matter). Nothing I do seems to work (which could be PEBKAC I appreciate)

lavolp3, 2018-05-27:
Can you share your config.txt and your cmdline.txt?

paulcarroll, 2018-05-28:
Absolutely   (and thanks for taking a look!).....

CONFIG.TXT
# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on
enable_uart=1
# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

CMDLINE.TXT
dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo

closed 2018-09-28 by goodtft