goodtft/LCD-show #30: Touch is not working on LCD 7" (B) 800x480

hasanvaez, 2017-07-22:
Hi and thank you for your efforts.
Unfortunately WaveShare does not have any driver for Touch on PI 3.
I have Raspbian Jessie 2017-07-05 (Latest Edition), But I could not run Touch for my LCD.
I have followed your instructions, but the problem is still remained. I have emailed to WaveShare support as my LCD is from WaveShare, but they did not answer me, yet.

Do you have any idea?

Best regards,
Hasan

goodtft, 2017-07-28:
Hello,
our Driver is newest update.
Our drivers will automatically load different configurations based on the Raspbian version to achieve maximum compatibility.
You can USE our Driver for test.
2017-07-05,we have tested on it, Touch and display all OK.
2017-07-28 

goodtft

goodtft, 2017-08-10:
Hello,
7inch Touch do not need to install any driver.
just Check your USB-wire is that OK ?or change a  good quality USB-cable for test
2017-08-10 

goodtft

closed 2018-09-28 by goodtft