goodtft/LCD-show #31: RTIMULib prob with Raspberry PI

faisal349, 2017-07-22:
Hi,

I am using RTIMULib with PI 3 for MPU9250 but unable to read anything. When I run the below code, it says 'No IMU found'. 

**RTIMUSettings *settings = new RTIMUSettings("RTIMULib");

    RTIMU *imu = RTIMU::createIMU(settings);

    if (imu == NULL || (imu->IMUType() == RTIMU_TYPE_NULL)) {
        printf("No IMU found\n");
        exit(1);
    }**

I used **i2cdetect -y 1** command to see if IMU is detected and I received 0x68 address.

Any ideas what I am doing wrong?

Thanks

faisal349, 2017-07-25:
I was using MPU9255 chip so I changed the address of MPU9250_ID to 0x73 in RTIMUDefs.h and it worked. :)

faisal349, 2017-07-25:
I have another question. I want to measure the position (x,y,z) so is there any routine in RTIMULib which I can call to get the position? I have looked through the library but couldn't find any routines for position.

Thanks

goodtft, 2017-07-28:
which LCD we are saying?

2017-07-28 

goodtft