goodtft/LCD-show #32: Can you please provide the same for different Distros?

shankreations, 2017-07-31:
Hey there, first thing first, **THANK YOU** so much. I just bought a 3.5 inch display for my RPi 2 B+ model, and was searching from continuously 18 hours for the supportive Drivers for the same, then got your GitHub Link, which helped me alot. Thank You again.

But, I mainly need it to work on **KALI LINUX (ARM version)**. On which, this driver is not working. _Can you please provide me the Driver for other Distro too (mainly for _**KALI**_)?_ or I would be glad if you can help me with the same in other way, like the dependencies it requires, repositories, tolls, etc. Thank you once again. Looking forward for your reply.

goodtft, 2017-08-24:
for KALI LINUX,you should USE Adafruit's Driver.
we also provide the kali image which have been istalled the driver. 
see ：
http://pan.baidu.com/s/1dF9yHlZ
PWS:z80w

texadactyl, 2017-10-17:
http://pan.baidu.com/s/1pL8ysDP ==>
"Oh oh, the page you are visiting does not exist."
Where is the Adafruit driver?