goodtft/LCD-show #34: Screen rotation

deanfantastic, 2017-08-22:
Hi
I've tried editing /boot/cmdline.txt and /bott/config.txt but neither are affecting a screen rotation of the GPIO 3.5" LCD screen with touch. The HDMI output does rotate but I want the LCD to rotate the same way.

How can I make this work as I'm very out of my depth on this.

Thanks
Dean

goodtft, 2017-08-24:
How to modify the display orientation
1.	GPIO interface type LCD rotating display method:
Suitable for GPIO interface type LCD (2.4” 2.8”3.2” 3.5”).
Run command at Terminal:
sudonano /boot/config.txt
if you are using 2.4”, 2.8”or 3.2”, then find line “dtoverlay=tft9341”and add rotate parameter value as format following：
dtoverlay=tft9341:rotate=value
if it is LCD 3.5”, then find line “dtoverlay=tft35a” and add rotate parameter value as format following：
dtoverlay=tft35a:rotate=value
Take LCD 3.5”as example:
To rotate 0 degrees (vertical screen display), the corresponding parameter value:
dtoverlay=tft35a:rotate=0
To rotate 90degrees (horizontal screen display), the corresponding parameter value:
dtoverlay=tft35a:rotate=90
To rotate 180degrees (vertical screen display), the corresponding parameter value:
dtoverlay=tft35a:rotate=180
To rotate 270degrees (horizontal screen display), the corresponding parameter value:
dtoverlay=tft35a:rotate=270
Press Ctrl + X, choose Y, save and exit; and re-start to achieve..

deanfantastic, 2017-08-24:
Does this also rotate the touchscreen as currently, I'm using:
`dtoverlay=piscreen,speed=16000000,rotate=270` which rotates the display but not the touchscreen

VivienGiraud, 2017-08-24:
@deanfantastic It won't works as it's not the same SPI interface, if I found a solution I will post it here :)

VivienGiraud, 2017-08-24:
These links might help you:
https://www.raspberrypi.org/forums/viewtopic.php?t=143581
https://ubuntuforums.org/archive/index.php/t-1493407.html
Please keep us in touch if it works, I don't have a piScreen so I can't test.

VivienGiraud, 2017-08-24:
Adding this:
```
        Option "InvertY" "true"
        Option "InvertX" "true"
```
To `/etc/X11/xorg.conf.d/99-calibration.conf`
Just before `EndSection`
I tried it with mine (Kuman 3.5") and it worked :D

deanfantastic, 2017-08-24:
Oh I'm not using a PiScreen, some kind of knock-off.

Tried all the above with SwapAxes and SwapXY but neither were effective.

```
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "MinX"  "3652"
        Option  "MaxX"  "62360"
        Option  "MinY"  "61728"
        Option  "MaxY"  "5342"
        Option  "SwapXY"        "1"
EndSection
```
and
```
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "MinX"  "3652"
        Option  "MaxX"  "62360"
        Option  "MinY"  "61728"
        Option  "MaxY"  "5342"
        Option  "SwapAxes"        "1"
EndSection
```

VivienGiraud, 2017-08-24:
Yeah it didn't works, you should try my solution.
Adding this:
```        Option "InvertY" "true"
        Option "InvertX" "true"```

VivienGiraud, 2017-08-25:
On my side it works flawlessly !

texadactyl, 2017-10-21:
Viv, please publish your config.txt setting for dtoverlay=tft35a parameter and the entire /etc/X11/xorg.conf.d/99-calibration.conf contents.

Thanks.

texadactyl, 2017-10-21:
Not intuitive at all.  The documentation is pathetic.  Sorry, goodtft, but it is true.

I started with a fresh raspbian install and LCD-show from github.  I stood the display up such that the power chord and HDMI adapter are pointing up (the way I want it when I deploy it).  Then:

=0 left
=90 bottom
=180 right
=270 top

i.e. counter-clockwise starting from the left.

It matters not what the contents of /etc/X11/xorg.conf.d/99-calibration.conf are.  I left the file as it was installed.

texadactyl, 2017-10-24:
My screen display orientation was solved.  But, my touchpad was inverted when I used one of my Tk applications and the desktop in general.  Viv's solution worked for me.

goodtft, your documentation sorely needs a refresh.

VivienGiraud, 2017-10-25:
@texadactyl Sorry for being late, really happy if my solution worked for you! You can close this one, and if you think documentation needs a refresh please open a new one with info related to.
Anyway I'm happy it works for you now, I'm really having funny with so, yeah! have fun with it too!

texadactyl, 2017-10-26:
deanfantastic was the originator and, therefore, should close it.  I don't know if this person has resolved their issues.

texadactyl, 2017-10-31:
Okay, I suspect that deanfantastic lost interest in the thread he started.  I'll try to close this issue.

moppymopperson, 2018-01-01:
@VivienGiraud and @texadactyl This helped me out a lot! Thank you for documenting your solutions!

dbvandy, 2018-01-29:
this saved my sanity... both fixes worked...

This setting in /boot/config.txt 
        dtoverlay=tft35a:rotate=270

and

/etc/X11/xorg.conf.d/99-calibration.conf
        Option "InvertY" "true"
        Option "InvertX" "true"

PERFECT! Thanks!

dromer, 2018-02-16:
There is no way to invert the x and y for the framebuffer?
I really don't want to use xorg, just with ie. pygame on a framebuffer.
plox

closed 2018-09-28 by goodtft

carpajr, 2019-04-30:
Is it possible to reopen this issue?

I'm trying to rotate touchscreen of tft35a using an eglfs application but didn't work. Even using EVDEV or TSLIB params.

texadactyl, 2019-04-30:
@carpajr If you want assistance, you have to provide more details than "didn't work".  WHAT is going wrong?  Rotated screen?  Unexpectedly zoomed in or out?  What specific remedies have you tried?

By the way, this vendor will provide no solution for you.  It's up to you to figure it out like everyone else in this stream of messages.  My messages describe what I did to make it work on an RPi.

carpajr, 2019-04-30:
@texadactyl, sorry. I've started to write and forgot to explain what happens.

I tried several ways to touchscreen work and these boot/config.txt lines were effective:
dtoverlay=tft35a:rotate=270
dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=1,pmax=255,xohms=60,xmin=200,xmax=3900,ymin=200,ymax=3900
dtoverlay=w1-gpio-pullup,gpiopin=4,extpullup=1

However, set up the screen rotation became a real challenge for eglfs applications. I tried several modifications in dtoverlay and decided to use ts_calibration, but also didn't work.

QT_QPA_FB_TSLIB=1
TSLIB_FBDEVICE=/dev/fb0
TSLIB_TSDEVICE=/dev/input/event0
TSLIB_CALIBFILE =/etc/pointercal

So, I tried to use ev_dev with some parameters, but once again, didn't work. The rotation configuration doesn't work.
QT_QPA_EVDEV_TOUCHSCREEN_PARAMETERS=/dev/input/event0:rotate=270 or 90

texadactyl, 2019-04-30:
My entire Raspberry Pi 2 /boot/config.txt after filtering out all of the comment lines:
```
dtparam=i2c_arm=on
dtparam=spi=on
enable_uart=1
dtparam=audio=on
dtoverlay=tft35a:rotate=270
```
Before on settling on rotate=270, I experimented with rotate values and observed the following:
```
	rotate=0           start of display area is at physical left
	rotate=90         start of display area is at physical bottom (up-side down)
	rotate=180       start of display area is at physical right
	rotate=270       start of display area is at physical top (right-side up)

```
I used the supplied goodtft program for calibration.  I did not hand-edit any calibration parameters in /etc/X11/xorg.conf.d/99-calibration.conf.  Here is the final file for me:
```
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "Calibration"   "3936 227 268 3880"
        Option  "SwapAxes"      "1"
    Option "InvertY" "true"
    Option "InvertX" "true"
EndSection
```

Then, sudo reboot.

Good luck.

texadactyl, 2019-04-30:
I have no file called "/etc/pointercal" on my RPi.  I am using Raspbian.  What are you using?
uname -a:
Linux rpiclock 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

My app is not based on QT.  Instead, I am using Tk/tkinter to access the TFT.

carpajr, 2019-04-30:
@texadactyl, 
I'm using Raspbian, but not using X11 or any windows system. The application that I'm developing runs over EGLFS (OpenGL) and that is why it is harder to do the touchscreen rotation. I didn't find any tip for fix this. 
I have not any problem using two other displays (7" resistive and 10" capacitive), but the tft35a seems a lost cause. :(

texadactyl, 2019-04-30:
You will have a challenge to train EGL and OpenGL in using goodtft hardware.  This TFT vendor supports only Xorg in combination with their custom Xorg code.

carpajr, 2019-04-30:
I didn't know about the limitation when I bought the display, but life is short to complain about that. :)

Yes, I saw the docs. There are some tricks to set up a touch, but none of them worked with tft35a.

alidevecioglu, 2019-06-12:
> Not intuitive at all. The documentation is pathetic. Sorry, goodtft, but it is true.
> 
> I started with a fresh raspbian install and LCD-show from github. I stood the display up such that the power chord and HDMI adapter are pointing up (the way I want it when I deploy it). This setting in /boot/config.txt worked for me:
> 
> ```
>      dtoverlay=tft35a:rotate=270
> ```
> 
> I observed:
> 
> rotate=0 top of screen is at physical left
> rotate=90 top of screen is at physical bottom (up-side down)
> rotate=180 top of screen is at physical right
> rotate=270 top of screen is at physical top (right-side up)
> 
> It matters not what the contents of /etc/X11/xorg.conf.d/99-calibration.conf are. I left the file as it was installed except following the touch-screen calibration instructions.

### It's working thank you 😃👍

fisher1977, 2019-07-02:
Option "InvertY" "true"
        Option "InvertX" "true"

It's working fine. thx

DjiZoN, 2022-01-14:
`    Option "InvertY" "true"
    Option "InvertX" "true"`

Thanks You Work Find. 

`    Option "SwapAxes" "1" or "0" `
this option for just accepted of invert orientation?????