goodtft/LCD-show #38: Xorg does not use 99-calibration.conf

drrnmoon, 2017-10-17:
So I calibrated my 3.5inch GPIO touch display (LCD35-show) and copied the result to ```/etc/X11/xorg.conf.d/99-calibration.conf```
but as I restart nothing seems to happen.
Everything is still the same. The touch is inverted and x and y are also inverted.
I tried hardcoding the /boot/config.txt by putting this code for the overlay:
```dtoverlay=tft35a:swapxy=1:xmin=22255:xmax=22255:ymin=49202:ymax=49612:yinvert=1:xinvert=1```
Still no result ...
Any suggestions?

texadactyl, 2017-10-24:
I started with a fresh raspbian install and LCD-show from github. I stood the display up such that the power chord and HDMI adapter are pointing up (the way I want it when I deploy it). This setting in /boot/config.txt worked for me:

     dtoverlay=tft35a:rotate=270

I experimented and observed:

rotate=0 top of screen is at physical left
rotate=90 top of screen is at physical bottom
rotate=180 top of screen is at physical right
rotate=270 top of screen is at physical top

Regarding orientation, it mattered not what the contents of /etc/X11/xorg.conf.d/99-calibration.conf were. I only modified the file for the purposes of calibration.

texadactyl, 2017-10-25:
To get the mouse-pointer to move in the same direction as the pen or a finger on the touch-screen, I added this just before EndSection in /etc/X11/xorg.conf.d/99-calibration.conf:

        Option "InvertY" "true"
        Option "InvertX" "true"

rosshadden, 2017-11-29:
Mine does not seem to use `99-calibration.conf` either. I had to do everything with the `dtoverlay` setting in `/boot/config.txt`.

rosshadden, 2017-11-29:
In my case I used `swapxy=1`, which worked, however I cannot invert the y-axis.  I have tried `touch-invy=1` which I found somewhere, and `invy=1`, and many variations. This is a nightmare. Any luck getting xorg to use the calibration file, @drrnmoon?

rosshadden, 2017-11-29:
I got it! The problem isn't that it's not using the file, it's that none of the properties in the file are supported anymore.  In more recent versions of Debian/Raspbian you have to set the transformation matrix of the touchscreen, using the `TransformationMatrix` property in the `99-calibration.conf` file.

Details:  https://github.com/notro/fbtft/issues/445

goodtft, 2017-11-30:
https://github.com/goodtft/LCD-show

2017-11-30 

goodtft

closed 2018-09-28 by goodtft