goodtft/LCD-show #39: Erroneous cp -rf used in LCD35-show bash script

texadactyl, 2017-10-18:
The `cp` -r option means recursive down a directory tree.  However, ./usr/99-calibration.conf-35 and ./usr/99-fbturbo.conf are being copied as individual simple files.  At minimum, this is confusing.  This may be okay now but it might bite you in the future when someone does maintenance.

The other bash scripts have similar issues.

texadactyl, 2017-10-18:
Minor point: Some indenting would make the bash scripts more readable.
E.g.
`If condition; then
        true-condition statements
else
        false-condition statements
fi`

lavolp3, 2017-10-26:
@texadactyl If you'd like you can have a look at my fork.
I've done quite some changes to the LCD35-show file which I'm trying to implement here one by one with pull requests (I'm new to this github stuff)
Have also tried to implement the several suggestions here.

goodtft, 2018-09-28:
been added in latest driver

closed 2018-09-28 by goodtft