goodtft/LCD-show #40: Debian package files (.deb) are a mystery

texadactyl, 2017-10-18:
Please create an area where the contents of the DEB files and the scripts which build them are available for browsing.  Thank you.

texadactyl, 2018-07-28:
Still no response.
Buyer beware.

danielhjames, 2019-08-05:
@texadactyl I don't see .deb packages in this repo any more. if you have .deb packages please make them available, as the control files could be extracted and would be a useful starting point for building current packages.

texadactyl, 2019-08-05:
@danielhjames I believe that I found a better software fall-back from this vendor while keeping the hardware: use Adafruit's solution for all of the RPi TFT's that they sell: https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/easy-install-2

I have not tried this "easy install, version 2" yet as I am busy with Astrophysics and Cryptography projects at the moment and my RPi is a lower priority.  But, I trust that Adafruit had success because of their proven expertise, my history with them, and knowing that they wouldn't publish something that they haven't tried themselves. 

One aspect that might be an issue: Adafruit might not be cognizant of the goodtft vendor.

danielhjames, 2019-08-06:
@texadactyl that's a very helpful link, thanks! As far as I can see, the LCD-show scripts only install one .deb package, which is a modified version of xserver-xorg-input-evdev, and not all screens require this .deb package. Scripts installing this package are:
```
LCD24-3A+-show
LCD24-show
LCD28-show
LCD32-show
LCD35-show
LCD5-show
MHS32-show
MHS35-show
MHS40-show
MPI3508-show
MPI4008-show
```

@goodtft it would help users if we understood the modifications made to the package xserver-xorg-input-evdev_1%3a2.10.5-1_armhf.deb compared to the upstream version. Could the modified package source be made available in a GitHub repository please?

Debian buster has a later version of xserver-xorg-input-evdev which is 1:2.10.6-1. It might make maintenance easier if we could upstream these modifications.

texadactyl, 2019-08-06:
@danielhjames I take your point about the DEB file contents and I know how to extract/build DTB files.  But, we choose our customizations, yes?  I think that I have invested enough time with this vendor. 

I try to make it a habit to stick with vendors who modify as little of the O/S as possible, write readable documentation, and are behave responsively.  Unfortunately, this vendor fails all those points.  Live and learn.

texadactyl, 2023-01-18:
No word from the vendor after 5+ years.

closed 2023-01-18 by texadactyl