goodtft/LCD-show #41: screen display white 3.5"

gazala123, 2017-10-18:
i have followed the EXACT CORRECT steps but when it reboots it still displays as white....

SHelfinger, 2017-10-21:
I had the same issue, even after I took the HAT off, my Pi didn't start.

Used card reader to replace all files (see in LCD-hdmi), connect the Pi in Network and use VNC to setup higher resolution again. I'll fork this Repo this week and change some stuff, I took a look in the files and realized some cmd are executed BEFORE check of "official" Rasbian version. Which means if you got "this is not raspberrypi kernel, no need to update touch configure, reboot" you screwed, since it just test if the Pi network name is raspberry, if you change the hostname you got a problem.

lavolp3, 2017-10-25:
It does indeed not make very much sense to check for the hostname. 
I forked this and tried out a check for the machine infrastructure (uname -m), which would result in armv6l or armv7l respectively dependant on the RasPi you use. From my point of view that would make more sense.
Will do a pull request soon.

dotps1, 2018-02-03:
this is absolutely ridiculous.  as soon as I change my hostname from raspberrypi, it no longer works.  how in the hell did anyone ever, EVER think this was a good idea?  I have three rpi's around I'm not going to leave them all named _raspberrypi_!

lavolp3, 2018-02-03:
I have changed this in my fork. There the if-statement does not look for
hostname but for architecture (raspi is arm6 or arm7). Not ideal since it
could be any arm6 but I guess it does the trick for this community and for
90% of the cases. Would be interested in a better suggestion.

Cheers