goodtft/LCD-show #42: Question: minecraft-pi black screen

jmkhael, 2017-10-24:
Hello,

After connecting the screen and flashing 3.5inch-raspbian jessie with pixel_2017-03-02.img and starting minecraft-pi, I get a black screen. Have you tried that? Is this supposed to work?

thank you

texadactyl, 2017-10-26:
Me too.  I suspect this is an issue with the minecraft-pi package with respect to a small-ish screen.  Pity!

I suggest that you write a detailed bug report on the raspbian site.

jmkhael, 2017-10-26:
righy. in fact minecraft-pi isn't compatible with this screen. GL problem. You'll have to live with an hdmi screen for minecraft...

texadactyl, 2017-10-27:
I have been doing a lot of Tk development lately, including testing with this LCD. I'd guess that it may just be a size issue. GL might be innocent. I have reproduced the black screen and other symptoms with my own apps inadvertently. 

Hard to tell. Games are complex, even a simple game like minecraft-pi.

lavolp3, 2017-10-27:
At the least it might be impossible to play Minecraft pit because of the
slow frame rate with this screen. Like discussed several time with regards
to retropie, the gpio connected screen will give you only 5-10 fps. Much
more seems to be impossible. Missing GL support using  gpio might be
another reason. Maybe the black screen had something to do with that.