goodtft/LCD-show #46: Plastic case breaks too easily

texadactyl, 2017-10-26:
I needed to pull off the case.  I was removing it quite gently with a spudger (aka black tip) when that piece of the case cracked.  I have never seen such a brittle piece of plastic!

texadactyl, 2017-12-02:
No response, I guess.
Buyer beware.

texadactyl, 2018-07-28:
Still no response.
Buyer beware.

stheil, 2018-11-15:
> Still no response.
> Buyer beware.

And there will probably never be one, seeing as you're complaining about a physical hardware issue on a software repository. Have you tried contacting the seller's customer support?

texadactyl, 2018-11-15:
The manufacturer is in Taiwan.  Their "customer support" is unresponsive.  I bought the hardware and initial software on Amazon; you can see my review there.

The manufacturer refers to this github project (goodtft) as their software support repository.  Unfortunately, it is not 100% open source (E.g. *.dtb).  The supplied documentation is extremely poor.  

I have left complaints in this github partly as a stimulus to the hardware product provider and partly to warn potential customers that this is not a simple hobby project.  

Current status: My github project (rpi_clock) based on the hardware product and this github is still functioning fine, somewhat because of my Unix/Linux experience.  If the github drivers suddenly stop working do to incompatibility with future Raspbian updates (quite possible), then I'll look for another solution for my github project.

MrEdinLaw, 2018-11-15:
> The manufacturer is in Taiwan. Their "customer support" is unresponsive. I bought the hardware and initial software on Amazon; you can see my review there: https://smile.amazon.com/gp/product/B06W55HBTX/ref=oh_aui_search_detailpage?ie=UTF8&psc=1#customerReviews (Richard).
> 
> The manufacturer refers to this github project (goodtft) as their software support repository. Unfortunately, it is not 100% open source (E.g. *.dtb). The supplied manufacturer documentation is extremely poor.
> 
> I have left complaints in this github partly as a stimulus to the hardware product provider and partly to warn potential customers that this is not a simple hobby project.
> 
> Current status: My github project (rpi_clock) based on the hardware product and this github is still functioning fine, somewhat because of my Unix/Linux experience. If the github drivers suddenly stop working do to incompatibility with future Raspbian updates (quite possible), then I'll look for another solution for my github project.

I bought a holder case (a case that holds internal parts) and the rasp 3.5tft screen for 12$ on ebay about a year or two ago, you just found a bad seller. Anyway good luck with your project mate.

texadactyl, 2018-11-15:
It's been done for a while.  Thanks for the good wishes.  

I am probably soon going to do a 2nd version of my rpi_clock project without this github site software as I heard from other Linux developers that Raspbian now supports TFT 3.5-inch screens.  It is certainly worth trying as the Raspbian drivers are all open source along with the rest of Raspbian.

texadactyl, 2018-11-15:
Just located this Raspbian-compliant tool from Adafruit:
https://cdn-learn.adafruit.com/downloads/pdf/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi.pdf

It uses only software in the Raspbian base and sanctioned 3rd party repositories.
Will give this a go when I get a break in my programming schedule.