goodtft/LCD-show #47: Orange PI Support for Touchscreen

alorbach, 2017-11-05:
Hi,

I have got the 5 Inch Display to work with changing HDMI Parameters in my fex config: 

> [disp_init]
> disp_init_enable = 1
> disp_mode = 0
> screen0_output_type = 3
> screen0_output_mode = 5
> screen1_output_type = 3
> screen1_output_mode = 5
> fb0_format = 0
> fb0_width = 800
> fb0_height = 480
> fb1_format = 800
> fb1_width = 480
> fb1_height = 0
> hdcp_enable = 1
> hdmi_cts_compatibility = 0
> 
> [hdmi_para]
> hdmi_used = 1
> hdmi_power = "vcc-hdmi-18"
> hdmi_x = 800
> hdmi_y = 480
> hdcp_enable = 0
> hdmi_cts_compatibility = 1

However I cannot get the touch screen to work. 
The Software that came with the screen is only for Raspberry, but I want it to work with Orange PI. 

Is there any hints how to get it running with GPIO or USB? Both methods do not work at the moment. 
I am using latest ARMBIAN on my Orange PI.

alorbach, 2017-11-05:
Would be great to know which drivers needs to be compiled within the kernel in order to get touch screen working.

oldxiao, 2017-11-19:
Hi alorbach,

i have installed Raspbian in my orangepi lite. but LCD incomplete display. could u put your orangepi's image on here? how to compile touch screen within the kernel. tks!

MasuodSamarin, 2018-11-29:
is there a chance to connect the 7 inch wareshare lcd to the orange pi one?