goodtft/LCD-show #48: Kali Linux Support

brunopava, 2017-11-09:
So here is the setup:

I downloaded the ARM image RaspberryPI Foundation from:
https://www.offensive-security.com/kali-linux-arm-images/

Then I proceed to mount the image to a SD card and boot the system.

On HDMI on my tv I then tried to install the LCD 5" driver following these steps:
`git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/ 
sudo ./LCD5-show `
As shown in the PDF that came with the CD for the Driver installation.(attached)

The system rebooted and the screen adjusted.
The actual LCD 5" HDMI display works just fine!

BUT, the system rebooted in text mode and I cant get it to work in graphical mode again.

I tried so far:

`startx`
This presented me with an error saying that 
(EE) Server terminated with error (1). Closing log file.log" for additional information.
(I can't find the log file for additional info, I don't know where it is)

`apt-get update
apt-get upgrade -y
apt-get install -f gdm3
gdm`
This did not work, the terminal was unresponsive after.

Do you guys have any idea what could possibly be going wrong?

[How to install the LCD driver (En).pdf](https://github.com/goodtft/LCD-show/files/1457581/How.to.install.the.LCD.driver.En.pdf)

goodtft, 2017-12-29:
We have  provide a Kali-Linux IMAGE  in CD.
you can used it for tested  directely, do not need to install any drivers.

brunopava, 2018-01-02:
Hi!

Where can I find this image?

mandelake, 2018-03-05:
For the touchscreen, you _(I didn't)_ might need an extra module - **ads7846_device.ko** - that is not a part of the kernel proper.

Download it from here: [https://drive.google.com/file/d/0BzIaxMH3N5O1c1c1b3ltNHFXRk0/view](url)

Add this to **/etc/modules**:

```
# WaveShare SpotPear 3.5" LCD
fbtft_device debug=3 rotate=90 name=flexfb speed=16000000 gpios=reset:25,dc:24
flexfb  width=320 height=480 regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0xF8,-1,0x11,-1,0x29,-3

# Touchscreen
ads7846_device model=7846 cs=1 gpio_pendown=17 keep_vref_on=1 swap_xy=0 pressure_max=255 x_plate_ohms=60 x_min=200 x_max=3900 y_min=200 y_max=3900
```

**Note!**   

_**- The forum might fold the lines. The line starting with "flexfb" should end with "-3". The last line should begin with "ads7846_device" and end with "3900"**._

Reboot.

Install the extra module for the touchscreen. Go to the directory where you saved the file and

```
sudo mkdir -p /lib/modules/$(uname -r)/extra
sudo install ads7846_device.ko /lib/modules/$(uname -r)/extra/
sudo depmod -a
```

Reboot.

Add this to the existing **/etc/X11/xorg.conf**

 ```
    Section "Device" 
    # WaveShare SpotPear 3.5", framebuffer 1
    Identifier "uga" 
    driver "fbdev" 
    Option "fbdev" "/dev/fb1" 
    Option "ShadowFB" "off"
    EndSection 

    Section "Monitor" 
    # Primary  monitor. WaveShare SpotPear 480x320
    Identifier "WSSP" 
    EndSection 

    Section "Screen" 
    Identifier "primary" 
    Device "uga" 
    Monitor "WSSP" 
    EndSection 

    Section "ServerLayout" 
    Identifier "default" 
    Screen 0 "primary" 0 0 
    EndSection
```

If for any particular reason **/usr/share/X11/xorg.conf.d/99-fbturbo.conf** conflicts with the above, you can delete it. 

Run `startx`. The desktop environment should load now on your touchdisplay with touch working as well.

KaosCreator69, 2018-03-21:
mandelake great steps, by any chance do you know what I would edit to make it work for 5" TFT touch?

Was able to get it displaying in Kali would love to have the touch screen working.

mandelake, 2018-03-24:
I wish I could help you, but I can't. The steps described above are for a [3.5" LCD module](https://ae01.alicdn.com/kf/HTB1IOptJXXXXXanXpXXq6xXFXXXz/144134313/HTB1IOptJXXXXXanXpXXq6xXFXXXz.jpg), because this is  what I have and could play with it.

brunopava, 2018-07-14:
@KaosCreator69 Hey, could you share your process please?
I'm still having some trouble and I'm kinda of a noob to figure it out by myself.

brunopava, 2018-07-14:
GUYS! 

I found this driver for 5" Kali Linux:
https://www.elecrow.com/wiki/index.php?title=HDMI_Interface_5_Inch_800x480_TFT_Display

Works perfectly! (ignore step 2)

closed 2018-09-28 by goodtft