goodtft/LCD-show #49: rotate lcd screen and touch

Sho0ter, 2017-11-20:
i would like to rotate my 7" lcd screen i did it 
but i have a probeleme with touch screen 
can i find  a solution ?

sarah278, 2017-12-17:
For the 3.5 TFT following change to /etc/X11/xorg.conf.d/99-calibration.conf did the job:

After 
        Option  "SwapAxes"      "1"
insert
        Option  "InvertX"       "1" 
        Option  "InvertY"       "1"

binaryjam, 2018-01-01:
I have done this but my touch is still not rotated.

I changed boot/config.txt
    dtoverlay=tft35a:rotate=90

(Although, the screen has always been landscape so this seemed unecessary, or there is other config)

I edited the calibration files and did all three of the above, and variations on them but the touch is always the same, x=y y=x.

binaryjam, 2018-01-01:
Ok I found a document that seems t work for anyone else with this problem
https://wiki.ubuntu.com/X/InputCoordinateTransformation

closed 2018-09-28 by goodtft