goodtft/LCD-show #54: Black screen after reboot

SamuellCline, 2017-12-08:
I have a 3.5" GPIO display, and after following the instructions, I get nothing but a black screen with a flashing underscore. The display is partially working, as I can press CTRL+ALT+F1 and see the terminal for just a moment, but it simply flashes on the screen and disappears.

eddymidget, 2017-12-28:
I'm having the same issue with the same screen, followed all the instructions and getting the same screen. CTRL+ALT+F1 however is not working for me

goodtft, 2017-12-29:
Which system version are you using for testing?

goodtft, 2017-12-29:
sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show

eddymidget, 2017-12-29:
The latest version of Raspbian

goodtft, 2017-12-29:
We have provided IMAGE with drivers.  You can use it for test:
链接：https://pan.baidu.com/s/1kVjBdXH 密码：0suf

goodtft, 2017-12-29:
3.5inch_raspbian20171129.7z
Raspbian Version:2017-11-29

eddymidget, 2017-12-29:
I can't download that because it needs a login with chinese number and I am from the UK?

dannienuc, 2017-12-30:
I just had this same issue after updating/ upgrading. The fix for me was to:

sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show
sudo ./LCD35-show

It appears that one of the changes breaks the driver included with the image you provide(not unusual with Linux). @goodtft I would recommend your next manufacturing run you include this possibility in the documentation.

graham768, 2018-01-13:
I'm still having this issue on the RPi zero with the same version of Raspbian as above.

graham768, 2018-01-13:
I downgraded to 2017-07-05 Raspbian and it works fine with the steps above. something in the latest version must cause the hangup

SamuellCline, 2018-01-13:
I also had to use an older version of raspbian to get it to work, unfortunately something broke in 2017-08-17.

o1gres, 2018-01-14:
> I downgraded to 2017-07-05 Raspbian and it works fine with the steps above. Something in the latest version must cause the hang up.

@graham768 I've the same problem with raspberry pi 0, where do you have downloaded the correct version of Raspbian? Thx

Barriol, 2018-01-20:
Thanks guys ! Two weeks trying to figure it ou what's wrong with my LCD display. Works fine on my RP3 but not my RPA+. Both works with older rasbian version.

KlaidKuzz, 2018-01-25:
Is this method still working? I tried installing the drivers on 2017-07-05 Raspbian but after a restart I'm stuck at mounting boot.

graham768, 2018-01-26:
As far as I know, this method still works on my RPi zero. You might want to create a new issue.

Roboblunder, 2018-06-04:
The problem still exists.  I had to revert to the July 5 2017 Raspbian to get it to work as well.  The manufacturer should fix this because I have other things that need an updated OS leaving me with no alternative but to ditch the screens and use another product.

pablommf, 2018-11-04:
Thank You man!

Eskimododd, 2019-04-20:
I am using this screen with Retropie on a RPi 3 B+. I entered the commands - 
```
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show
sudo ./LCD35-show
```
When it reboots, I get a white screen, then the boot menu come up on the 3.5 screen, but then Retropie defaults to the HDMI output. If I go back to command prompt, that comes up on the 3.5 screen. 

Any ideas how I get Retropie to display on the 3.5 screen? Thanks in advance

albfan, 2024-11-11:
I can confirm latest version (at this moment  October 22nd 2024) from

https://www.raspberrypi.com/software/operating-systems/

and last version of LCD-show (in my case LCD35-show)

start showing a splash screen, then a black screen (with an underscore char at top-left)

Ctrl+Alt+Fx works randomly (screen alternates from tty login to this black screen with an underscore)

My guess is at some point raspberry moves to wayland, but not really sure.

As ssh works perfectly you can track journalctl -xf or similar but there's nothing obvious on logs.

Currently install 2017-07-05 raspbian version from

http://downloads.raspberrypi.org/raspbian/images

works if you use the matching version from LCD-show at that moment:

3bc8e597bfb2dc53b8f2a24186a4118591290109

there's no option to rotate screen at that moment, and I would say calibration is inverted (x is y, and direction is reversed), so I would start a bisect from this commit to latest version and find what final config is done to identify changes and see if last version of LCD-show applies same config or different

albfan, 2024-11-16:
@farhan123mohd instructions are there, so just install that raspbian version 2017-07-05 and checkout that LCD-Show version:

```
git clone https://github.com/goodtft/LCD-show
cd LCD-show
git checkout 3bc8e59
./LCD35-show
```

if your LCD version is exact as mine, you can use my fork to fix the touchscreen:

```
git clone https://github.com/goodtft/LCD-show
cd LCD-show
git checkout 2017-35-bugfixing
./LCD35-show
```

Hopefully someone will detect what is different lately and make it work with recent raspberry OS

farhan123mohd, 2024-11-17:
Thank you sir, but i couldn't install 2017 version. But instead i installed
tge new legacy versuon whi h works fine.

albfan, 2024-11-17:
That indeed is great feedback.

https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/

One of most importsnt changes from Bulleye (legacy) to Bookworm is changes from X11 to wayland.

Makes sense this repo (all focused on xorg technology) doesn't work on wayland oriented OS.

So I think this repo just need a PR stating It needs to use x11 oriented OS,  to close this issue

gitqrr, 2024-12-14:
So are there any way to use this 3.5 inch display without downgrading RPi OS?

albfan, 2024-12-14:
Probably following this issue instructions:

https://github.com/goodtft/LCD-show/issues/369

That basically disabling wayland to use X11

But that itself is another downgrade.

 Cannot find info on this TFT screens working on wayland

Probably direct cable to show image is the solution people use