goodtft/LCD-show #58: Change resolution on 3.5 lcd-gpio

nesmon, 2018-01-11:
I have a 3.5 lcd conected via a gpio port but the resolution is to small.
How change this for a beter experience ?

Sorry for my english, im french

lavolp3, 2018-02-02:
Go into the /boot/config.txt and find the line below

`hdmi_cvt 480 320 60 6 0 0 0`
The first and second value (480 and 320 in this case) define resolution. 480 x 320 is the highest the 3.5 inch tft can generate natively. I wouldn't recommend a higher resolution.

nesmon, 2018-02-02:
Ok thanks a test this

nesmon, 2018-02-03:
The problème with this it's my screen doesn't use hdmi. 
It's a only gpio screen

lavolp3, 2018-02-03:
Usually if you have a 3.5'' gpio lcd display it is natively 320x240 or
320x480. Higher resolutions are not supported and not advised.  It would
not display well. If you just use the script to install the driver the
native resolution will be used.
What (more) do you need?
What is the display you are using?

jiortigosa, 2018-03-31:
Hello,
I am trying to work with only the 3.5" LCD display with GPIO.
No HDMI monitor attached to the Raspberry PI 2.

I have "updated" the LCD-show drivers to the latest version of 3 days ago (end of March 2018).

I get the LCD to display a part of the complete screen, and I have not been able to "reduce" the resolution of the Raspbian screen under 640*480 pixels...

I looked at my /boot/config.txt file, and I have not found the line you mention above.

I am attaching my file /boot/config.txt:
[config.txt](https://github.com/goodtft/LCD-show/files/1865597/config.txt)

I would appreciate your help very much. I am not a Linux command line expert...

closed 2018-09-28 by goodtft