goodtft/LCD-show #60: Kernel panic-not syncing: VFS: unable to mount root fs on unknown- block(179,2) running Raspbian on top of NOOBS

PagesByZ, 2018-01-13:
I already have Raspbian through NOOBS installed and whenever I install the LCD-show driver, i get the following error during reboot after installation:

     `Kernel panic-not syncing: VFS: unable to mount root fs on unknown- block(179,2)`

Any idea what is the issue and how I can fix it? I am installing 2017-11-29 version of NOOBS.

lavolp3, 2018-01-13:
The reason is that NOOBS has the rootfs in another partition.
Mount your sd card in your pc, open the cmdline.txt with an editor (not
Word or Wordpad!!) and change the term after rootfs (should be mmcblk0px)
to mmcblk0p7.

wizzahd, 2018-01-22:
thanks @lavolp3, this fixed the issue for me!

Dj-Ale-426, 2018-02-02:
Hi i am new to raspberry pi. I am wondering where do you find cmdline.txt in the files

Dj-Ale-426, 2018-02-02:
@lavolp3 i just found the cmdline.txt but there is no rootfs and mmcblk0px

lavolp3, 2018-02-02:
The cmdline.txt is in the boot directory and can also be editet in Windows (using notepad or notepad++)
What do you find in there. Can you paste it.

Another way around this issue would be to setup the system again from scratch and use my fork [here](https://github.com/lavolp3/LCD-show).
I have included a NOOBS case creating the right cmdline.txt and avoiding the kernel panic

Dj-Ale-426, 2018-02-03:
@lavolp3 thanks for helping me

dotps1, 2018-02-03:
@lavolp3 I see you have a lot of code commented out at the bottom of https://github.com/lavolp3/LCD-show/blob/master/LCD35-show, can it be used if I use your repo rather then this one?

lavolp3, 2018-02-03:
> LCD-show/blob/master/LCD35-show, can it be used if I use your repo rather

The reason is I don't understand  what the if statement does. From my
understanding the version check  and resulting commands were necessary for
older raspbian wheezy versions but are not anymore now.
So I changed it so the necessary commands get executed w/o the check for a
version.
I'm not 100% sure it works so I just commented out the old code without
deleting it. Please try it out if you like.

lavolp3, 2018-02-03:
@dotps1 The code works on my Pis as it is.

Dj-Ale-426, 2018-02-06:
@lavolop3 yes I can paste in code. If it is not a bother can you type how the cmdline.txt is supposed to look so I can paste it in.

Dj-Ale-426, 2018-02-10:
@ lavolp3 yes I can paste in code. But if it is not bother can you type how the codling.txt is supposed to look like.

lavolp3, 2018-02-10:
@Dj-Ale-426 
dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo

Basically the same cmdline.txt as in this git but for noobs you need mmcblk0p7 to be the root partition instead of mmcblk0p2

Dj-Ale-426, 2018-02-10:
@lavolp3 the thing is I tried your git but it just displayed kernel  panic not syncing

lavolp3, 2018-02-28:
@Dj-Ale-426 Sorry for the long silence.
Please, point out, what exactly you did:
Which Raspbian did you use (jessie, strech? lite or full?)?
What did you do to install the display? Step-by-step please.

Dj-Ale-426, 2018-03-01:
Hi sorry I had noobs installed so I quickly fiqured ou the problem. Thanks for your concern. Also is it okay if I use the lite version of raspian

Dj-Ale-426, 2018-03-01:
Also for noobs can you just adjust the cmdline.txt

crashflow, 2018-03-02:
This fixed it for me. Try this first.

Dj-Ale-426, 2018-04-03:
@dotps1 will this work for  noobs lite too.

antonisss, 2019-05-09:
Hello!!! I am new to raspberry. I have the same problem Kernel Panic: VFS : Unable to mount root fs on unknown-block 179,2 but when i open the cmdline.txt didnt find any mmcblk0p7 in there. Please help !!!

lavolp3, 2019-05-09:
can you post your cmdline.txt here?

NedAckland, 2019-05-09:
i have a very similar problem;
cmdline.txt - dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=43c1e3bb-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh splash plymouth.ignore-serial-consoles

version - April 2019 -Raspbian Stretch with desktop and recommended software
kenel version - 4.14

first time using a raspberry pi, using b3+, didn't want to get noobs because of the size.

lavolp3, 2019-05-09:
Are you booting from an external device?

antonisss, 2019-05-09:
no i am not booting from external device

NedAckland, 2019-05-09:
booting from the pi

lavolp3, 2019-05-09:
Well I don't know where it is from but the root=UUID hints to a bit from an external partition.
For sure it is the reason why the pi won't boot. Google it. My suggestion : overwrite the /boot /cmdline.txt with the one from this repository  /usr/cmdline.txt.
Make a backup of the original one first. Boot and see what happens. If it boots consider running the LCD-show script again.

NedAckland, 2019-05-09:
thank you, will try this

antonisss, 2019-05-09:
will try

DUANBEN, 2019-05-19:
@NedAckland hi, i have the same problem as you, have you sovled this trouble?

NedAckland, 2019-05-19:
i did solve it, as long as you dont have data on the sd, you can fully over
write it using sd card formatter (i did the full over write rather than
then quick one) then using etcher i re put rasbian on the sd and booted it
up again. this is what worked for me in the end.

DUANBEN, 2019-05-20:
i tried it this way, but it didn't work for me....

thank you anyway :)

rprior04938, 2019-05-25:
I had the same problem and managed to fix it.  I want to explain the situation and the fix.  The situation occurs because of the interaction of Noobs and The LCD-show scripts.  The problem is the SD card partitions which are set up by Noobs versus those set up if you put Raspbian onto an SD card without Noobs.    The basic setup of operating systems disks (SD card or other) on the Raspberry Pi is to have 2 partitions - a boot partition and a root partition.  If Raspbian is alone, the boot partition is partition 1 and the root partition is partition 2.  With Noobs, the boot partition of the first-loaded operation system is #6 and the root partition is #7.  The LCD-show scripts hard-code the root partition to be 2.
The fix to this problem is mostly explained above.  You will need an operating computer OS and an SD card reader-writer.  You can use the Pi itself if you have a 2nd SD card with a working Linux or Raspbian OS and a USB SD card reader.  If you use OSX (for the Mac) or Linux, you can use it without extra software.  If you use Windows, you'll need Cygwin or other software to read and write the non-Windows filesystem ext4.
You need to understand the partitioning scheme of the SD card.  Typically, Noobs sets up 2 partitions per OS starting with partition6, and installs Raspbian onto partitions #6 (boot) and #7 (root).  You will need to "mount" Raspbian boot partition using the CLI of Mac/Linux "mount" command and to use an editor such as nano to edit the single line of cmdline.txt, changing the "root=/dev/mmcblk0p2" to "root=/dev/mmcblk0p7", or other "mmcblk0px" where x is 1 more than the Raspbian boot partition.
Then umount the SD card, put it into the Pi, and reboot.

varshan97, 2020-02-23:
hey i'm trying to install Kali in rasp pi Zero w, it's showing "end kernal panic not sync: VFS: unable to mount root fs on unknown-block(179,2)

irfan12345678, 2020-05-18:
> can you post your cmdline.txt here?

Hello sir I am new to raspberry. I have the same problem Kernel Panic: VFS : Unable to mount root fs on unknown-block 179,2 but when i open the cmdline.txt didnt find any mmcblk0p7 in there. Please help !!! please help me

rprior04938, 2020-05-18:
See my posting from 2018 in the GitHub discussion.  The part of the line in
cmdline.txt (found on the boot partition of the SD card, typically
/dev/mmcblk0p6, the 6th partition of the card if the SD card was created by
NOOBS) created by NOOBS, is /dev/mmcblk0p7, the 7th partition of the card.
It points to the Linux root partition.  If the root partition is some other
place, it can't be found and you get a kernel panic.

If you don't have access to another way to access the Linux partitions of
your SD card, I'd suggest that you start over with a fresh download and
burn to the card of Raspbian or other pi Linux OS.

jn7779, 2021-05-17:
`Kernel panic-not syncing: VFS: unable to mount root fs on unknown- block(179,2)`
how can i fixed this error with my cmdline.txt?
console=serial0,115200 console=tty1 root=PARTUUID=566d1a16-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

ChrBlad, 2021-12-20:
> 

Did you find a solution?

ChrBlad, 2021-12-20:
> `Kernel panic-not syncing: VFS: unable to mount root fs on unknown- block(179,2)` how can i fixed this error with my cmdline.txt? console=serial0,115200 console=tty1 root=PARTUUID=566d1a16-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Did you find a solution?

jn7779, 2021-12-20:
I have try some of the solution, but she like my SD card might be corrupted.