goodtft/LCD-show #63: Black screen on Raspberry Pi Zero W

kabongsteve, 2018-01-27:
Unable to get it to work on a Raspberry Pi Zero W

Get the black screen like other issue raised.

Trying older version doesn't seem to work either.
Image from CD worked, but no WiFi and don't know how to enable
Says "no wireless interfaces found"

Trying to use raspian-Stretch

anyone able to help?
Thanks

pfeerick, 2018-04-12:
A good start to working out what was wrong here is what display you are using... as there are several supported by this repo... the 5" HDMI display I have had some issues, and may have been related to the below lines as it seemed to go away when I commented them out (on the rPi Zero W with latest raspbian stretch)

```
hdmi_force_hotplug=1
config_hdmi_boost=7
```

granpino, 2018-05-22:
I have a 3.5" HDMI screen and it works great with raspberry pi 2. However with the raspberry pi 1B, I also get a black screen after boot.