goodtft/LCD-show #64: GUI not loading after doing given steps on Kali Linux

techifi, 2018-01-27:
I did everything as per the steps given and the display started working but after that the GUI is not loading and only large script is running and big terminal is showing up nothing more then that after login.

scolandrea, 2018-02-08:
I've the same issue. Does anyone give us a hand here? Kali GUI is broken after ran the provider script.

Thanks!

lavolp3, 2018-02-08:
So what is exactly happening? Don't you get some error message? Then please  paste it here

scolandrea, 2018-02-08:
Thanks for reply,

In my case I don't get a any message. I ran the script exactly how is describe in the documentation, after the first reboot GUI doesn't work anymore I always get the command line with login when I boot my raspberry. This only happens with Kali I've another SD with a RetroPie and the screen works perfectly fine.

I would like resolve this issue I don't how if I need do something else.

I hope you can help me,

Thanks again!

theotuck, 2018-07-10:
I get this problem too. 
After running script LCD7C-show , only the tty works but no GUI. 
LCD-hdmi  just changes the  resolution ratio but not the GUI.
I tried the cmd  [`startx`], but error occurs.  **fail to load module "ftturbo"**.

then I `mv the 99-ftturbo file to ~` , then the `startx` works, I got a GUI.  However, my usb devices, such as mouse and keyboard can work as well. So I have to build my kali system.

GDXFR, 2018-08-23:
Hi, have you solved this issue?

theotuck, 2018-08-25:
yes.
I change the default setting for screen, restart the system then the problem solved.

closed 2018-09-28 by goodtft