goodtft/LCD-show #65: Raspbian Lite/7 inch type b (800x480) unable to connect to X server

adsmz, 2018-01-27:
On a Pi 3 running raspbian Lite, I'm unable to use startx after installing the driver for the 800 by 480 7 inch screen. Xorg log: http://ix.io/EMD

lavolp3, 2018-01-28:
Raspbian lite usually does not have a guide installed. Then startx does not
work.
You could either use raspbian full or install the x server in your lite
installation afterwards, e.g. with apt-get install xserver-xorg lxde

adsmz, 2018-01-28:
I already had xserver-xorg installed, but removed and installed it again to be sure. It still doesn't work

lavolp3, 2018-01-29:
To me there seems to be a problem with the fbturbo driver.
Can you check if the script really has copied the 99-fbturbo.conf to /usr/share/X11/xorg.conf.d/99-fbturbo.conf?

Additionally, I would try installing the fbturbo driver as described here:
`https://github.com/ssvb/xf86-video-fbturbo/wiki/Installation`
But I'm not sure why this would be important. Maybe it's missing when you "post-install" the xserver. 
At the moment I'm not even sure why the fbturbo driver is needed at all...

adsmz, 2018-01-29:
Installing fbturbo as described in that link  worked, thanks

closed 2018-01-29 by adsmz

doskoi, 2018-08-12:
Thanks a lot, my waveshare LCD was work with X, but after install the LCD-show to install touch support.
the X isn't working anymore, after I try to install fbturbo, the X is back with touch support.