goodtft/LCD-show #66: Stream pi camera to LCD

haykp, 2018-02-01:
I installed the code, works perfectly!!!
Thank you Author for it, very much

The only issue i have is showing the pycamera data right on LCD.
Can you please help to resolve that issue also?

Now when I run pycamera, it makes preview on HDMI port.

Thanks

lavolp3, 2018-02-02:
Please see point 5.4 in the picamera [FAQ](https://picamera.readthedocs.io/en/release-1.13/faq.html)

> 5.4. The preview doesn’t work on my PiTFT screen: The camera’s preview system directly overlays the Pi’s output on the HDMI or composite video ports. At this time, it will not operate with GPIO-driven displays like the PiTFT. Some projects, like the Adafruit Touchscreen Camera project, have approximated a preview by rapidly capturing unencoded images and displaying them on the PiTFT instead.

All displays using SPI will be too slow for a picamera preview since they are physically limited to around 8 fps.
You won't have much fun using one of these.
As mentioned there seems to be a workaround by adafruit. They usually have a good documentation

haykp, 2018-02-15:
Thank you
I tried adafruit, seems working so far

closed 2018-09-28 by goodtft