goodtft/LCD-show #67: Error installing driver "cp: cannot create regular file ‘/usr/share/X11/xorg.conf.d/99-fbturbo.conf’

codyvogt, 2018-02-06:
When installing the driver, i get this error cp: cannot create regular file ‘/usr/share/X11/xorg.conf.d/99-fbturbo.conf’: No such file or directory when running this command sudo ./LCD7B-show

lavolp3, 2018-02-06:
What OS do you have installed?
Is the x-server installed?

codyvogt, 2018-02-06:
I think i figured it out. I ran this sudo apt-get update && sudo apt-get install xserver-xorg-video-fbturbo before installing and it seemed to work. I was trying to install on a Retropie image.
Thanks!

closed 2018-02-06 by codyvogt

Chip-Tronics, 2019-04-09:
I'm having the same issue, I tried your fix, and the display still doesn't work. This is the 2nd display, as I thought the first one was broken.
Is there anyone who can put down the exact instructions they used.
This is really driving me nuts now. A whole with trial and errors.
Then I input the last part of the installation code, my pi drops the ssh connection and reads the same this as above. No such file.

sunnyspeedstudio, 2021-01-12:
> I'm having the same issue, I tried your fix, and the display still doesn't work. This is the 2nd display, as I thought the first one was broken.
> Is there anyone who can show the exact instructions they used?
> This is really driving me nuts now. A whole week with trial and errors.
> Then I input the last part of the installation code, my pi drops the ssh connection, pi then reboots, after the system shows the same thing above. No such file.

I had the same issue, and figured out.
Use the Raspbian with Desktop GUI version, not the lite version (without desktop GUI)

TalibaniGrannyFanny, 2021-01-13:
Yeah I have had the same problems, on my orangepipc, raspberry pi 0/RPi0W,
rpi3b+, too; loooong and frustrating story short, my only way of fixing
this was to get an osoyoo 3.5” hdmi screen instead; plug and play,
perfect....

Not strictly a’fix’, I know, but still...

If you happen to find a proper workaround fix for this could you please let
me know what it is please...??

WELEE666, 2024-01-14:
何解