goodtft/LCD-show #68: LCD35-show uses uname -n to determine if raspberry pi

mjf55, 2018-02-08:
The current version of LCD35-show ( and probably others scripts ) current uses uname -r to determine if the kernel is a raspberrypi.  

nodeplatform=`uname -n`
kernel=`uname -r`
version=`uname -v`
if test "$nodeplatform" = "raspberrypi";then

If not it will not check for updates.  However, uname -n actually returns the network node hostname.  Many folks, especially if you have multiple raspberry pis, will change the hostname to a network unique hostname to avoid confusion.  

So, using uname -n is not a reliable way to determine if your are indeed running on a pi.    

But, I do not have a solution for you.

lavolp3, 2018-02-09:
I have already sent a Pull request with a possible solution. You can use the infrastructure instead (armv6 or armv7). There may be other computers using this infrastructure but the solution is still far better. It's integrated in my fork.

mjf55, 2018-02-10:
Ok, thanks.  I just wanted to point it out.  I made my 'workaround' by changing nodeplatform test in the script.  
Isn't Pi2 an armv6  and Pi3 an armv7 ?  maybe test for either of them, and assume that it is connected to a Pi.  Anyway.  Thanks.

jantman, 2018-08-02:
The information found via ``/proc/cpuinfo`` is probably the most reliable way to check this; see https://elinux.org/RPi_HardwareHistory

goodtft, 2019-07-01:
We will solve this problem