goodtft/LCD-show #69: no image on 3.5 inch touch screen

9jobsworth, 2018-03-16:
every time there is an upgrade to raspian there seems to be a touchscreen problem.
still at least i can switch back to my standard hdmi output.
latest raspian stretch - 3.5 inch touch.
my screen did work with an old jessie image supplied
but on upgrade the screen does not work.
keep trying and thank you for the tips here.
john

lavolp3, 2018-03-18:
Does the screen output itself still work? Only touchscreen funktion gets broken?

With update, what exactly do you mean? `apt-get upgrade`? If yes, can you share your sources list?

Roboblunder, 2018-06-04:
Do not upgrade.  The driver only works with the July 5th 2017 release or older of Raspbian.

9jobsworth, 2018-06-04:
i have given up with this touch screen. it is old stock and i made a bad mistake to buy it on aliexpress.
as you say it only works with an old version of raspbian.
thank you all of you who contributed to this thread.

closed 2019-07-01 by goodtft