goodtft/LCD-show #70: Releases please

mpeck-me, 2018-03-18:
Please make releases so that it's easier to find downloadable copies of your code.  Github will create a zip/tgz of your source tree, so people don't need to mess around with git.

I went through the pain of installing cygwin, then adding git (I thought it was in the base devel set), and THEN was able to get this awesome code to install.  (Yes, I made it harder on myself installing cygwin, but I prefer cmd-line to the gui tools, and something that runs in cmd.exe I would claim is not actually a cmd-line tool.)

By the way, if you haven't gotten feedback about it, the version that is there now, the readme shows as:

2017-08-16 Update:
v1.6-20170824
Update xserver to support Raspbian-2017-08-16

works great with a Landzo 3.5" on 2018-03-13-raspbian-stretch.

Thanks, and good job!

goodtft, 2018-03-19:
The Release version you can Download on:
http://www.hellolcd.com/wiki/images/6/60/LCD-show.zip

mpeck-me, 2018-03-20:
That link, while it may work, is not useful.  It doesn't show a version and is not mentioned anywhere.

I strongly suggest the github releases model instead.  If you upload that zip file to hellolcd, great, but github is the source of truth.

pfeerick, 2018-04-12:
Why do you **need** releases to access it as a zip/tgz??? Use the "Clone or Download" button and choose the "Download ZIP" option... it gives you a zipped up version of the current branch, and you don't need to mess with gygwin or the github client. 

[image]

mpeck-me, 2018-04-12:
Honestly, I've been using Github professionally for nearly 4 years and didn't know about that.

Mentioning it in the notes would help people find that.

I had two reasons for filing this: 1. it wasn't documented how to get the software, for someone who doesn't know/want to use git; and 2. there (still isn't) a way to get a specific release.  Always taking top-of-tree can be dangerous/unreliable.  I understand that ToT probably isn't as dangerous for this package, however that's rather unusual.

Anyway, I think I've taken as much of people's time as this deserves; I wanted to give my 2 cents and you've done some of what I asked, so I'm happy with the results.

Thanks!!

Mike

closed 2018-09-28 by goodtft