goodtft/LCD-show #71: After install raspbain each time asks to press Enter for login

haykp, 2018-03-24:
After installing the driver, now each time when I boot the raspberry it brings following message:
"Cannot open access to console, the root account is locked. See sulogin(8) man page for more details
Press Enter to continue"

Only after pressing enter  the OS is loaded.

Does someone know how to get rid of this issue?

ldab, 2018-05-11:
Same problem here

rajkissu, 2018-06-08:
I had the exact same issue after installing the driver, this fixed it for me http://mapledyne.com/ideas/2015/8/4/reset-lost-admin-password-for-raspberry-pi

Do ensure you have an SD card reader and access to a laptop - my raspberry pi locked when I added `init=/bin/sh` to **/boot/cmdline.txt**, so I never to the part on resetting my admin password.

I had to remove it by editing the file externally on my laptop. Once I plugged the SD card back in, my pi worked as per normal again.

totial, 2018-10-03:
had the same problem and managed to fix it by "verifying" both boot and system partition with gparted on linux. 
Steps are: 
1) open gparted, right click -> unmount partition for both partitions on SD card.
2) right click on partition -> verify also for both partitions
3) apply and thats it. You should be ready to go