goodtft/LCD-show #72: Touch driver ads7846 disables /dev/spidev*: incompatibility with other SPI devices (RFID reader, can't open device)

fcorucci, 2018-03-30:
Hello, 

I am using these drivers in order to operate a 5" Kuman Touch LCD on a Raspberry PI 3 model B, running Raspbian 9 (stretch). Unfortunately I am experiencing compatibility issues among the touch display and other SPI devices. Let me briefly explain the issue.

In addition to the touch display, I also need an RFID tag reader (RFID-RC522 MFRC522) to operate over the same SPI bus. My understanding is that this should be possible in theory, provided that the two devices are wired correctly, which, I believe, I've done.

Now, the RFID reader works perfectly until I install the touch LCD drivers (i.e. until I run ./LCD5-show): after that, it doesn't work anymore ("can't open device: No such file or directory" "Aborted").

The error seems to be related to the fact that /dev/spidev* disappear after installing the touch driver.
Looks like this is due to the LCD drivers modifying /boot/config.txt to load the ads7846 overlay (my understanding is this is related to the LCD touch). This turns out to be disabling the spidev devices that are expected by the RFID reader.

My questions are:
- why does the touch overlay disable spidev?
- is there anything I can do to solve this issue and have the RFID reader working alongside the touch display?

Thanks

References regarding the RFID reader: 
http://www.instructables.com/id/Raspberry-Pi-3-Model-B-MIFARE-RC522-RFID-Tag-Readi/
https://github.com/lthiery/SPI-Py.git
https://github.com/mxgxw/MFRC522-python.git

Bogatinovski, 2019-01-04:
I have exactly the same scenario. Is there a way to use both the touchscreen and the RFID reader ?

jannisbaudisch, 2019-01-06:
See https://github.com/notro/fbtft/issues/282

I had the same issue. Unfortunatly I have no clue of how device trees works and the solution above didn't work for me. 

This was my solution:
1. decompile the ads7846 device overlay file with `dtc -I dtb -O dts -o /tmp/ads7846.dts /boot/overlays/ads7846.dtbo`
2. open the decompiled file with `nano /tmp/ads7846.dts` and change every line with `status = "disabled"` to `status = "okey"`
3. remove old device tree overlay `rm /boot/overlays/ads7846.dtbo`
4. compile your new device tree source file and save it in the overlays directory `dtc -I dts -O dtb -o /boot/overlays/ads7846.dtbo /tmp/ads7846.dts`

I have no idea if this is a good solution, but at least it works for me.

fcorucci, 2019-01-07:
In the end I've used a workaround by enabling an additional SPI bus - SPI1 - on Raspberry and moving one of the two devices to this bus. I am no expert in device trees related issues either, but from what I've learned about this problem I do agree that @jannisbaudisch's should work. I just hope it doesn't cause other troubles/weird interactions (there must be some reason why the ads7846 overlay explicitly disables the spidev device: maybe the touch controller doesn't work well in a multi-slave scenario, so the developers felt like preventing this by hiding the spidev device? Just speculating).

Cheers!

nizoo91, 2019-03-12:
@Fcorucci

Can you elaborate on how you got SPI1 working? 

To my understanding and from reviewing the files, it also disables SPI1, the only way to get SPI1 to show in /dev/ for me was to decompile, edit and recompile the .dtbo file(s) related to my specific screen....

However even than, it doesn't seem to work at all :/

fcorucci, 2019-03-13:
@nizoo91 in my case the touch driver only seems to disable spidev0.* devices, not the auxiliary spidev1.*

You can enable an auxiliary SPI interface (SPI1) with one chip select by specifying 
`dtoverlay=spi1-1cs`
This will enable the device `/dev/spidev1.0`

You can also add more chips selects (i.e. more devices) if you need to, e.g. this enables two of them:
`dtoverlay=spi1-2cs`
in this case you'll enable `/dev/spidev1.0` and `/dev/spidev1.1`

More info here:
https://elinux.org/RPi_SPI
https://librpip.frasersdev.net/peripheral-config/spi1-aux/

You can find the pinout of the auxiliary SPI device here: https://it.pinout.xyz/pinout/spi

I hope this helps!
Cheers

nizoo91, 2019-03-15:
got it finally work! 

Cheers @fcorucci & @jannisbaudisch 

Since I was using MHS35 file, it was actually disabling both of them so I ended up having to re-compile the file and activate SPI1! 

Thanks guys!

jatinpaliwal, 2019-04-21:
@fcorucci
can you provide the pin connections you used for rc522 have the same issue and need to use both the touchscreena nd the rc522 at the same time.