goodtft/LCD-show #76: How to uninstall?

JsBergbau, 2018-04-14:
After installing each time is asked for pressing enter after boot, thats ridiciolous. How can the whole thing removed that it is the same before without the display?

paulcarroll, 2018-05-25:
I'm only learning myself, but all I did  was  ssh into the pi and run "sudo ./LCD-hdmi"... everything back the way it was and hdmi working perfectly

jagij, 2018-07-10:
We encountered the same problem before. It wasn't related to the display though, something got corrupted on the sdcard.
The solution in our case was to check the drive for errors:
```
mount -o remount,ro /
fsck /
```
Then answer yes to all (unless you don't agree).
Shortcut:
```
mount -o remount,ro /
yes | fsck /
```
Don't forget to reboot afterwards!

tonb, 2020-08-31:
The install seems to write a system_restore.sh file which you can run to uninstall.

tyler-canton, 2021-06-22:
I'm confused why is there not an uninstall exe. Whats going on guys?

Tranks27, 2022-06-23:
> I'm only learning myself, but all I did was ssh into the pi and run "sudo ./LCD-hdmi"... everything back the way it was and hdmi working perfectly

i did "sudo ./LCD-hdmi". It rebooted but now none of them turns on. Please help

MartinRGB, 2023-09-30:
> I'm only learning myself, but all I did was ssh into the pi and run "sudo ./LCD-hdmi"... everything back the way it was and hdmi working perfectly

work for me