goodtft/LCD-show #77: Not working with google assistent

chathurangawijetunge, 2018-04-25:
I have installed google assistent on my pi and I have installed lcd 3.5.
The lcd work's fine till assistant starts after assistant starts screen goes white.

Why is That.
How can I use both together...?

Thanks

chathurangawijetunge, 2018-04-25:
When assistant virtual environment starts LCD goes wite

simoautio, 2018-05-01:
I have the voice hat v1, and my setup used to work fine before i recently updated the the system (new kernel version). When i run one of the assistant .py files, the mouse cursor starts to jump around the the touch gets kind of broken.

It could be just that the voice hat & touchscreen are both using one the same pins, but this used to work before the latest kernel update.