goodtft/LCD-show #78: Crash Volumio Os

qpens, 2018-05-06:
After installing I'm getting kernel panic

[image]

lavolp3, 2018-05-06:
You need to go into the /boot/cmdline.txt and find the rootfs entry.
Change the last number from 2 to 3.
Volumio creates three partitions, standard raspbian has only two.

qpens, 2018-05-06:
I made new Install.
My cmdline.txt:
dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo

After installing driver:

[image]

lavolp3, 2018-05-06:
Try changing
root=/dev/mmcblk0p2
into
root=/dev/mmcblk0p3

And then boot...
There's a lot on this issue if you google volumio and kernel panic

qpens, 2018-05-06:
[image]

qpens, 2018-05-06:
I have noticed that that script is replacing cmdline.txt and I made some changes instead raspberry I put volumio and put original cmdline.  And now Its dead without HDMI and still no screen on 3.5lcd ;(

this is orginal from volumio
splash quiet plymouth.ignore-serial-consoles dwc_otg.fiq_enable=1 dwc_otg.fiq_fsm_enable=1 dwc_otg.fiq_fsm_mask=0xF dwc_otg.nak_holdoff=1 console=serial0,115200 kgdboc=serial0,115200 console=tty1 imgpart=/dev/mmcblk0p2 imgfile=/volumio_current.sqsh elevator=noop rootwait bootdelay=5 logo.nologo vt.global_cursor_default=0 loglevel=0

lavolp3, 2018-05-07:
Why aren't you trying out what I suggested?

The Volumio installation also replaces cmdline.txt and you just used the installation-cmdline.txt and not the final one for the running system.
You can find the volumio cmd-line.txt [here](https://github.com/volumio/Volumio-WebUI/blob/master/_OS_SETTINGS/boot/cmdline.txt)
It reads `force_turbo=1 dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p3 elevator=noop rootfstype=ext4 rootwait smsc95xx.turbo_mode=N
`
...

qpens, 2018-05-07:
I try with that line and change in script raspberry to volumio and same kernel panic.  Some guy from volumio forum gave this info:
 
"Well, I still doubt cmdline.txt, because your screenshot shows that the kernel can’t find init, this means Volumio is not booting with its specific parameters in cmdline.txt"

lavolp3, 2018-05-26:
I will try to test that out for myself soon... also would like to try out volumio again.

qpens, 2018-05-26:
cool

lavolp3, 2018-07-06:
@brian777777 
I presume you mean cmdline.txt and not cmd-line.txt
It is IMO really very much advisable to only change the rootfs entry as shown above instead of the whole cmdline.txt. The file is very sensitive towards formatting.
Can you paste your new cmdline.txt here?