goodtft/LCD-show #80: where is the source code ?

flatmax, 2018-05-09:
I would like to look at the source code - where can I find it ??

lavolp3, 2018-05-26:
Here? Don't exactly know what you're looking for but it's all here I guess...

flatmax, 2018-05-28:
Just binaries here - right.
Where is the source code ?

lavolp3, 2018-05-28:
As far as I understand this is a specialized solution for a system of Raspberry Pi + Screen. Since it is mostly only a "correction" of already existing files (config.txt, cmdline.txt) and no driver per se, there is no source code for these available.

Regarding the .dtb overlays and the evdev files: These have just been put here, but have been developed somewhere else. You will need to look for the source code somewhere else.

surban, 2018-06-07:
@lavolp3 Do you know where exactly?

texadactyl, 2018-07-11:
Except for the /usr/*.dtb files, almost all of the other files are readable scripts and configuration files except for 3 DEB files in the tree top:
   xinput-calibrator_0.7.5-1_armhf.deb
   xserver-xorg-input-evdev_1%3a2.10.3-1_armhf.deb
    xserver-xorg-input-evdev_2.10.5-1_armhf.deb

I cannot figure out why this vendor would be including *.deb files since they are readily maintained in Raspbian and Debian.

The Device Tree Blob (DTB) files should be documented by this vendor.  An explanation of device blobs can be found here: https://www.raspberrypi.org/documentation/configuration/device-tree.md.

tesseract241, 2018-11-25:
Device Tree Blobs are a bit different from normal binaries, meaning that the compilation process does not actually do much more than erasing earlier redefinition of parameters and putting all node definitions together.  
As a result, you can easily decompile dtb files using the device tree decompiler, dtc like this:

`dtc -O dts -I dtb YOUR_BLOB_FILE`

You can install dtc through your repository tools, look for device-tree-compiler and dtc.

tl;dr: you can get decompile the blob easily, you're gonna get what's effectively but not technically the original source like that

texadactyl, 2018-11-25:
Agreed.  

I can also disassemble Java class files, Python pyc files, and compiled object files for forensic purposes.  Viewing the original source shows documentation at a detailed level as well as intent - useful for self-education as well as auditing.  I prefer the philosophy of "show your work".

tesseract241, 2018-11-25:
As I said before, dtb are not "compiled" in the sense object files are, and dts files are little more than a list of parameters with the right values.  
But you're right, comments would be deleted by the compilation process, so I guess you would lose those, if they happened to exist.

texadactyl, 2018-11-25:
I ran `dtc` on bcm2710-rpi-3-b.dtb on an RPi3.  The first 5 messages are warnings about directories that do not exist (E.g. /memory and /soc):

<stdout>: Warning (unit_address_vs_reg): Node /memory has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc/vchiq has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc/local_intc has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc/gpiomem has a reg or ranges property, but no unit name

Maybe their corresponding dts files are out of date or erroneously coded.  No comments so it is a challenge to guess their intent.

In any case, they should not be supplying hardware-related device definitions that are subject to update during an `apt` process.  I could understand X-Windows configuration changes based on specific hardware.

ericschultz, 2019-05-23:
Seriously, where is the original dtb sources? I'm not even sure it's legal for you to distribute them without the source.

Tianxiaomo, 2020-10-07:
So,where is the original dtb sources?
-_-???

cschweers, 2020-10-07:
for ARM devices(like Raspberry): https://github.com/torvalds/linux/tree/master/arch/arm/boot/dts

xversial, 2022-09-26:
Another vendor to avoid.