goodtft/LCD-show #81: Black screen raspberry pi 1B

granpino, 2018-05-22:
I have a 3.5" HDMI screen and it works great with raspberry pi 2. However with the raspberry pi 1B, I get a black screen after boot. I'm using the latest raspbian stretch. If I hash tag the last line in the config.txt the screen works with a black border around it but no touch.

L3gomancer, 2018-05-31:
Hi! It sounds like an overlay issue. Have you tried reinstalling?

I tried it but it was with Stretch Lite, that's Raspbian without desktop. these were the steps that worked for me.

Setup: Raspberry Pi 1 B [revision](https://www.raspberrypi-spy.co.uk/2012/09/raspberry-pi-board-revisions/#prettyPhoto) 2.0 ("2011.12" printed), with LCD, +TV, ethernet, keyboard, power
I burned Raspbian Stretch Lite to a 4GB SD card
Inserted in Pi, booted, connected
Installed Git:
$ sudo apt-get install git
and then LCD software:
$ git clone https://github.com/goodtft/LCD-show.git
$ chmod -R 755 LCD-show
$ cd LCD-show
$ sudo LCD35-show
After auto-reboot it showed on both TV and LCD

granpino, 2018-06-18:
The problem is with a HDMI 3.5 screen and MPI3508-show.  After I uninstall with LCD-hdmi, the screen returns to normal however when I connect a mouse the screen goes black again. If I disconnect the mouse the screen returns with the password box as if the screensaver has taken over. Is the mouse actvatng the screensaver? This is a fresh STRETCH installation. I will have to use the screen without the touch and use a mouse. Thanks