goodtft/LCD-show #83: Freezes if I am booting from usb flash disk

berk76, 2018-06-08:
Hello,
when I am booting form SD all is ok and works very well. 
But if I try to boot from usb flash disk it freezes.

Thank you for your nice project :-)

17Tango, 2018-06-17:
I have exactly the same issue...
Had running 2 Raspberry Pi´s with the 7" 1024x600 Touch LCD with SD-Card as boot device without any problems for months.
But 24/7 usage kills the SD-Cards soon. Now using 32GB SSD (Transcend) as boot device with latest version of Raspbian (3-2018) successfully as system, except after i tried to use the LCD´s again. Display works, but boot sequence freezes always at the same line.
Pi´s power supply is 100% (measured), as LCD and SSD get seperate 5V each.
Booting from SD still works well, but don´t wanna go back because SD isn´t reliable enough for my project.
Any help, hint or fix are welcome.
Thanks in advance for support!

kinga-mate-tech, 2018-11-26:
This can be fixed by altering the script or after installation taking the usb stick out and plugging it into a different system to modify the cmdline.txt in the boot partition.

You are looking for 

> root=/dev/mmcblk0p2
this needs to be changed to 

> root=/dev/sda2

This is because it is only referencing sd cards and not usb storage.

kinga-mate-tech, 2018-11-26:
Reference for the devs... 

https://github.com/balena-os/resinos/issues/396

17Tango, 2019-09-20:
Well, I know it's a few months ago. Actually was reading the reply earlier, but just tried it now.
Hell ya, it works like a charm now!

THX a lot @kinga-mate-tech !!!

17Tango, 2019-09-20:
Running even on Raspbian-Buster now, just need to run calibration tool for touch input.
sudo apt-get install -y xinput-calibrator