goodtft/LCD-show #92: Rotate the screen?

MarkoDamien, 2018-07-30:
Is there a possibility to rotate the screen? Like for 180 Degrees

jantman, 2018-08-03:
Yes, edit ``/boot/config.txt``.

Using the installer here, it will have a line in it like  ``dtoverlay=tft35a``

If you want to rotate 180 from the default, change that to ``dtoverlay=tft35a:rotate=270``

(note the default seems to be 90...)

MrEdinLaw, 2018-08-03:
I can confirm this works, Thanks

closed 2018-09-28 by goodtft

MrEdinLaw, 2018-10-10:
> Yes, edit `/boot/config.txt`.
> 
> Using the installer here, it will have a line in it like `dtoverlay=tft35a`
> 
> If you want to rotate 180 from the default, change that to `dtoverlay=tft35a:rotate=270`
> 
> (note the default seems to be 90...)

This seems to rotate the screen but not the touch. The touch seems to stay the same

andrew-rbd, 2019-04-26:
How about for the mhs35? I tried:
dtoverlay=mhs35:rotate=90
and it does not work.

goodtft, 2019-04-29:
Is there any printed MHS-3.5inch printed on the back of the 3.5inch screen you purchased?
If there is no such silk screen, then you are using ordinary version module, instead of dtoverlay=mhs35 instruction, you use tft35a-overlay. DTB

2019-04-29 

goodtft

goodtft, 2019-04-29:
Try：
dtoverlay=mhs35:rotate=0

2019-04-29 

goodtft

andrew-rbd, 2019-04-30:
MHS-3.5inch Display is printed on the back of the display.

_____

Andrew Bryant  |  Founder & CEO
Right Brain Development
 
695 East McGlincy Lane Suite A
Campbell, CA 95008

office 1-408-840-2003 ext. 102
mobile 1-408-348-2173
 
https://rightbraindev.com