goodtft/LCD-show #93: Rotating Touch [Raspbian 4.14, 3.5 LCD, RP 2]

auxua, 2018-08-05:
Did anyone manage to get the touchscreen running correctly on an existing Raspbian?

The Display is working, but the touch inputs are rotated by 90°

I tried:
  * Calibrating as in the documentation
  * Swapping Axis in all configs
  * Swapping axis manually in /etc/modules
  * Setting Transformation matrices in xinput manually

goodtft, 2018-08-06:
http://www.lcdwiki.com/7inch_HDMI_Display-C

2018-08-06 

goodtft

AustinSaintAubin, 2018-08-12:
I am having the same issue. @auxua where you able to resolve this?

AustinSaintAubin, 2018-08-12:
Here is what I did to get it to work.
`Linux 4.14.61-v7+ #1133 SMP Fri Aug 10 11:04:43 BST 2018 armv7l GNU/Linux`
`LCD-Show v1.7-20180320`

--------

```
sudo -i
```

```
# https://www.raspberrypi.org/forums/viewtopic.php?t=162323
# https://www.raspberrypi.org/forums/viewtopic.php?t=148678
cat > /etc/X11/xorg.conf.d/98-lcd-show.conf <<"EOF"
Section "Device" 
	# WaveShare SpotPear 3.5", framebuffer 1
	Identifier "uga" 
	driver "fbdev" 
	Option "fbdev" "/dev/fb1" 
	Option "ShadowFB" "off"
	#Option "Rotate" "UD", Flip Upside Down
	Option "SwapbuffersWait" "true"
EndSection 

Section "Monitor" 
	# Primary monitor. WaveShare SpotPear 480x320
	Identifier "WSSP" 
EndSection 

Section "Screen" 
	Identifier "primary" 
	Device "uga" 
	Monitor "WSSP" 
EndSection 

Section "ServerLayout" 
	Identifier "default" 
	Screen 0 "primary" 0 0
EndSection
EOF
```

```
# https://www.waveshare.com/wiki/7inch_HDMI_LCD_(C)
cat > /etc/X11/xorg.conf.d/99-calibration.conf <<"EOF"
Section "InputClass"
	Identifier      "calibration touchscreen default primary uga fbdev"
	MatchProduct    "ADS7846 Touchscreen"
	Option  "Calibration"   "3936 227 268 3880"
	Option "CalibrationMatrix" "0 -1 1 1 0 0 0 0 1"   # 270 degree
	#Option "CalibrationMatrix" "0 1 0 -1 0 1 0 0 1"  # 90 degree, Flip Upside Down
	###Option	"SwapAxes"  "1"
	###Option	"SwapXY"	"1"
	###Option	"InvertX" "true"
	###Option	"InvertY" "true"
EndSection
EOF
```

`reboot`

auxua, 2018-08-22:
Did not work for me. Still no change. Retried from scratch without success - Bought another LCD from another manufacturer and it worked without problems

goodtft, 2018-09-28:
please check the following url: http://www.lcdwiki.com/res/Show_Direction_and_Touch/How_to_change_display_direction-GPIO-Resistive_Touch-V1.2.pdf
and feedback this work or not

auxua, 2018-10-11:
Sorry, Pi broke - cannot test for some time

closed 2018-10-17 by goodtft