goodtft/LCD-show #96: Touch screen is backwards

nickm324, 2018-08-25:
Installed on Raspbian Stretch and display works but touch screen is backwards and flipped. If I touch and move up and down the cursor moves left and right, if I move left and right it moves up and down.

And the cursor location when touching the screen is opposite where you touch. If I touch screen in upper left corner the cursor jumps to bottom left hand corner.

MrEdinLaw, 2018-09-07:
Having the same problem here.

nickm324, 2018-09-07:
I did come up with a solution to this issue but I would have to look at my Pi to see what I did exactly if you are still having the issue let me know and we can compare our configs. When I get time I will post my solution here anyway in case someone else has the same problem.

mtbsteve, 2018-09-08:
same issue here. Pls let me know what you have done to resolve Thanks!!

mtbsteve, 2018-09-09:
OK I finally found the solution. 
Since Rasperian “Jessi” version the kernel / img uses libinput as it's mouse drivers instead of the old evdev, which  ignores 99-calibration.conf swapaxis options.
The LCD touchscreen drivers as provided on this git still relies on evdev. 
See here how to resolve  it:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=179477#p1163460

@goodtft can you please fix this in the code. Thanks!

andriesruiter, 2018-09-11:
On a raspberry pi make sure the hostname is exactly raspberrypi. Eg your prompt is pi@raspberrypi 
If not, correct it in raspi-config under network options.

After this, run sudo ./LCDxx-show again

At least this worked for me with a raspberry pi 3b and a Velleman VMP400 screen.

nickm324, 2018-09-12:
Unfortunately this has nothing to do with the hostname.

goodtft, 2018-09-28:
Hi all,
please help to check if this problem still happened in the latest driver on github. Thanks!

netty2018, 2018-10-06:
> Installed on Raspbian Stretch and display works but touch screen is backwards and flipped. If I touch and move up and down the cursor moves left and right, if I move left and right it moves up and [...]

lcd_rotate=2 is the solution.  sudo nano /boot/config.txt
Add one of the following lines to the bottom of the file :

display_rotate=0
display_rotate=1
display_rotate=2
display_rotate=3
0 is the normal configuration. 1 is 90 degrees. 2 is 180 degress. 3 is 270 degrees.
If you are using the Official Raspberry Pi touch screen you can use “lcd_rotate” rather than “display_rotate”.
Save the file by using CTRL-X, Y then ENTER.
http://osoyoo.com/2016/05/26/osoyoo-lcd-touch-screen-for-raspberry-pi-installation-guide/

and lastly you will need this part http://www.circuitbasics.com/raspberry-pi-touchscreen-calibration-screen-rotation/   its all reading and maybe a 10 min vid to watch you will have it done in 30 min :)

etherialice, 2018-10-09:
> 
> 
> Hi all,
> please help to check if this problem still happened in the latest driver on github. Thanks!

I just got my mpi3508 display and downloaded the driver today, and I'm getting this issue

goodtft, 2018-10-17:
> > Hi all,
> > please help to check if this problem still happened in the latest driver on github. Thanks!
> 
> I just got my mpi3508 display and downloaded the driver today, and I'm getting this issue

which OS are you using(UBUNTU,KALI,RASBIAN....)? we can not reproduce this issue.

goodtft, 2018-10-20:
http://www.lcdwiki.com/3.5inch_HDMI_Display-B

2018-10-20 

goodtft

Dtrieb, 2018-12-26:
To solve the Problem:

`sudo apt-get install xserver-xorg-input-evdev`
`sudo cp -rf /usr/share/X11/xorg.conf.d/10-evdev.conf /usr/share/X11/xorg.conf.d/45-evdev.conf`

Ricardo-Osorio, 2020-07-02:
All I had to do to fix this for me was edit the file `/etc/X11/xorg.conf.d/99-calibration.conf` and change the line that said invert axis to 0 (false/off)

80avin, 2021-11-06:
As of 6 November 2021, I can confirm @Ricardo-Osorio 's comment. All I required was playing with "SwapAxes"  and "rotate.sh" command in this repo

80avin, 2021-11-06:
I can confirm @Ricardo-Osorio 's comment that as of 6 Nov 2021, all I needed was changing "SwapAxes" in `/etc/X11/xorg.conf.d/99-calibration.conf` to 1 and `sudo ./rotate.sh 180`