goodtft/LCD-show #99: windblows IOT riot

netty2018, 2018-10-06:
oh my what a failure this thing is a nightmare to install on a pie if i solve it i will post

netty2018, 2018-10-07:
down vote
favorite
1
Following my previous question, I now have a working LCD display, but with a different issue.

The problem is the same as in this question, yet I am unable to find a solution.

I don't have a xorg.conf file, but in xorg.conf.d/99-calibration.conf I put the result of xinput_calibration, which is:

Section "InputClass"
    Identifier  "calibration"
    MatchProduct    "ADS7846 Touchscreen"
    Option  "MinX"  "61348"
    Option  "MaxX"  "3095"
    Option  "MinY"  "3738"
    Option  "MaxY"  "62309"
    Option  "SwapXY"    "1" # unless it was already set to 1
    Option  "InvertX"   "1"  # unless it was already set
    Option  "InvertY"   "1"  # unless it was already set
EndSection
xinput list gives

⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ ADS7846 Touchscreen                       id=6    [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
xinput list-props 6 gives

Device 'ADS7846 Touchscreen':
    Device Enabled (116):   1
    Coordinate Transformation Matrix (117): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
    libinput Calibration Matrix (246):  1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
    libinput Calibration Matrix Default (247):  1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
    libinput Send Events Modes Available (248): 1, 0
    libinput Send Events Mode Enabled (249):    0, 0
    libinput Send Events Mode Enabled Default (250):    0, 0
    Device Node (251):  "/dev/input/event0"
    Device Product ID (252):    0, 0
    libinput Horizontal Scroll Enabled (253):   0
and, as you can see, none of the options specified in 99-calibration.conf is used.

If I try running, as suggested in the answers on the other question

xinput set-prop 6 "Evdev Axis Calibration" <Min-X> <Max-X> <Min-Y> <Max-Y>
xinput set-prop 6 "Evdev Axes Swap" 1

goodtft, 2018-10-09:
what problem with you production?

netty2018, 2018-10-11:
its fine now it has some strange script in it i couldn't identify it said it was split i guess i must have copied a link into it by accident that wasn't compatible THANKS

closed 2018-10-17 by goodtft