goodtft/LCD-show #100: Turn Display off and on using console/ script?

markus-hsk, 2018-10-07:
If I type `vcgencmd display_power 0`, the display signal is gone as expected, but the display itself stays on with a blue screen, saying that there is no signal:

[image]

Is it possible to switch the screen off completely to save power?

goodtft, 2018-10-09:
there's no way to turn off backlight by using console, but you can turn off backlight by long press(3 seconds) the backlight key

mtraves, 2018-11-20:
Where might this "backlight key" be?  

Won't the screen become damaged by staying online indefinitely?

cstodd, 2018-12-30:
Would love it if it had a “backlight key”, but it doesn’t. Have the 3.5 model. Is there really no way to turn off the backlight via software or otherwise?? If so, returning this immediately and leaving a negative review.  That’s basic functionality.

withakay, 2019-01-18:
@goodtft is there is any way to turn of the back light on the 3.5" display?

cechevarne, 2019-01-18:
@goodtft is there is any way to turn of the back light on the 3.5" display?

wixaw, 2019-03-14:
Hello
I just bought the "3.5inch_RPi_Display".
How can I turn off the backlight and turn it on when I touch the screen
If only I had knowledge of this git before....

dioro-zz, 2019-08-16:
I have the same question... Can I turn off the backlight and turn on via command script or tourchscreen ?

Vertigo1206, 2019-11-15:
On the display with "backlight switch" maybe you can try to use a transistor, optocoupler or similar parallel and switch this one with GPIO.

andreas-globi, 2020-09-25:
I was struggling with this too. Using `vcgencmd` to turn the display off leaves the LCD on and bright blue (yuck).

I found a good cheat though. Since the display is getting it's power from the Pi's USB ports, you can simply turn off the USB power on the Pi using uhubctl (https://github.com/mvp/uhubctl)

To turn off: `sudo /home/pi/uhubctl/uhubctl -l 1-1 -p 2 -a 0`
And back on: `sudo /home/pi/uhubctl/uhubctl -l 1-1 -p 2 -a 1`